chipsalliance/rocket-chip #14: Convert the scala sources of Zscale(changed) to verilog in rocket-chip.

huangchengxtu, 2015-08-14:
Hi everyone, and thanks for your work.
I have changed the Zscale. Mainly is to change the bus. I want to Convert the scala sources of Zscale i changed to verilog in rocket-chip. I only need Zscale. Using the Zscale i changed replaces the zscale in Rocket-chip, and then "make CONFIG=ZscaleConfig MODEL=ZscaleTop verilog", but Something is wrong. I do not know how to config.  Do me need to change other place? as makefile or some configuration,   then convert the scala sources of Zscale i changed to verilog in rocket-chip. 
thanks for your help!

sbeamer, 2015-08-18:
This appears to be asking the same question as #12. What is wrong with the verilog produced by `make verilog`?

sbeamer, 2016-01-06:
Please comment again to re-open this issue.

closed 2016-01-06 by sbeamer