chipsalliance/rocket-chip #21: Unable to compile using make in vsim directory

nitish2112, 2015-10-19:
Hi,

 I am trying to make vcs simulator but getting the following errors:

Warning-[ERASM] Argument size mismatch
/ufs/vlsi/nks45/RISCV/rocket-chip/vsrc/rocketTestHarness.v, 396
"htif_fini(1'b1);"
  External routine argument size mismatch.
  Use +lint=ERASM-L for details.

50 modules and 2 UDPs read.
	However, due to incremental compilation, no re-compilation is necessary.
make[1]: Entering directory `/ufs/vlsi/nks45/RISCV/rocket-chip/vsim/csrc'
make[1]: Warning: File `filelist.cu' has modification time 2 s in the future
ld -shared  -o .//../simv-DefaultVLSIConfig.daidir//_csrc0.so objs/udps/zTfCi.o objs/udps/iMDw8.o 
rm -f _csrc0.so
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsGroupGetActive'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `miHeapProf_init'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `printIclBeginMarker'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCallocFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `executeTclProc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckMallocFromGroupFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `waitForUserInput'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `storeSimLogFileHandle'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsSetOutOfMemoryRoutine'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `registerPreRestartFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `initNativeUcli'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckReallocFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsNamedAllocateGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `printIclEndMarker'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCallocFromGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsHighMemory'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsStrdupFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckMallocFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsGetMemBytes'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsMallocFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsReallocFromGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckCallocFromGroupFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `execMhpiTclCommand'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsMallocFromGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsSetHook'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsFreeFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsMemSetLogFileHandle'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsFreeGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsOutOfMem'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsNamedPushGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `miHeapOn'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `setIclCmd_ResVars'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `setEnvPair'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsLowMemory'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckFreeFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `registerOutOfMemHandle'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `vcsRunUcliErrorScript'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `shutDownNativeUcli'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsMemManagerIsActive'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsDoStackUnwind'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `breakNativeTclLoop'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckCallocFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsNamedPopGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `storeMiscTfCb'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCurrentGroup'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `isUcliEngineStarted'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsPrintf'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckStrdupFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsucli.so: undefined reference to `execMhpiTclCommandNoPrint'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsGetHook'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsReallocFunc'
/ufs/cad/synopsys/vcs-K-2015.09/linux64/lib/libvcsnew.so: undefined reference to `snpsGroupSize'
collect2: error: ld returned 1 exit status
make[1]: *** [product_timestamp] Error 1
make[1]: Leaving directory `/ufs/vlsi/nks45/RISCV/rocket-chip/vsim/csrc'
Make exited with status 2
make: *** [simv-DefaultVLSIConfig] Error 2

palmer-dabbelt, 2016-03-22:
We can build with VCS versions close to that one, so I think it's probably a configuration issue on your end.  If you have more info, feel free to re-open this, but I think SolvNet might be the right place to get a fix.

closed 2016-03-22 by palmer-dabbelt

kkiningh, 2017-04-01:
I had a similar problem. The relevant Solvnet article suggested that I had conflicting versions of VCS installed, which was not a problem in my case.

The issue for me was that gcc by default uses slightly different linker options on Debian/Ubuntu than on Fedora (which is what VCS is tested on). To fix this it I added -LD_FLAGS -Wl,--no-as-needed to the list of vcs options. That means a full command would look something like

vcs -full64 +v2k top.v -LDFLAGS -Wl,--no-as-needed
For more details on this see this SO post.

haseebazaz-lm, 2020-06-15:
> I had a similar problem. The relevant Solvnet article suggested that I had conflicting versions of VCS installed, which was not a problem in my case.
>
> The issue for me was that gcc by default uses [...]

Is there a way I can resolve this problem permanently because I have to add this line in Makefile all time. I am using ubuntu 18.04