chipsalliance/rocket-chip #26: GLIBC version used

nitish2112, 2015-11-05:
Hi, 

  I am having some issues while compiling the design using VCS. The errors are coming from the g++ invocation by VCS. Can you please tell which version of glibc was used for the project. The one I am using is:

%ldd --version

ldd (Ubuntu EGLIBC 2.15-0ubuntu10.12) 2.15
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

 Thanks a lot :)

ben-k, 2015-11-11:
Looks like we are (typically) using glibc version 2.19.

colinschmidt, 2016-02-11:
Re-open this if you are still having an issue. Please include the error log from g++ failing to compile when/if you do.

closed 2016-02-11 by colinschmidt