chipsalliance/rocket-chip #44: Chisel is being invoked too many times on error (3x)?

ccelio, 2016-02-11:
I'm trying to generate C++ emulators, and if I have a Chisel compile error, I get Chisel being invoked three times (I'm expecting Chisel to be invoked twice, the first to attempt to compile the hardware quickly, and the second to generate the proper stack trace as required if an error is found). 

I'm wondering if the emulator/Makefile may be the culprit:

````
44 $(generated_dir)/$(MODEL).$(CONFIG).d $(model_header) $(params_file): $(chisel_srcs)                   
45    cd $(base_dir) && $(SBT) "project $(PROJECT)" "run $(CHISEL_ARGS) --noIoDebug"                      
46                                                                                                        
````

Is anyone else seeing this behavior?

palmer-dabbelt, 2016-02-22:
This is because there's no "%" in the list of targets of this recipe.  A non-pattern recipe with multiple targets is actually just multiple recipes, so this is directly telling make to run Chisel three times.  You'll probably also see this fall over and die on parallel builds.

The easiest way to fix it is probably to just have a Chisel stamp, but finding a way to massage this into a pattern-based recipe would also fix the problem.

palmer-dabbelt, 2016-03-22:
I believe I've fixed this in the Chisel 3 Makefile, so I'm going to close the issue since the old version will eventually go away.

closed 2016-03-22 by palmer-dabbelt

ccelio, 2016-03-22:
Good enough for me!