chipsalliance/rocket-chip #50: Error compiling emulator "make -j8 run-asm-tests"

davidfongsj1108, 2016-02-22:
Hi,

I was able to compile the emulator with no error

cd emulator
make debug

I get this error related to 

riscv64-unknown-elf-gcc: error: ./../env/entry.S: No such file or directory
and it's probably related to the installation.

Which make or build cmd-line will install the relevant files under
/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/env

Thanks,

David

make -j8 run-asm-tests-debug
(installation echoed out to screen)
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-simple.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-add.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-addi.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-and.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-andi.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-auipc.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-beq.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-bge.hex
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
riscv64-unknown-elf-gcc  -DENTROPY=19162 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/simple.S -lc -o rv64ui-v-simple
riscv64-unknown-elf-gcc  -DENTROPY=19164 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/andi.S -lc -o rv64ui-v-andi
riscv64-unknown-elf-gcc  -DENTROPY=19166 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/auipc.S -lc -o rv64ui-v-auipc
riscv64-unknown-elf-gcc  -DENTROPY=19169 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/bge.S -lc -o rv64ui-v-bge
riscv64-unknown-elf-gcc  -DENTROPY=19168 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/and.S -lc -o rv64ui-v-and
riscv64-unknown-elf-gcc: error: ./../env/v/entry.S: No such file or directory
riscv64-unknown-elf-gcc: error: ./../env/v/vm.c: No such file or directory
riscv64-unknown-elf-gcc  -DENTROPY=19163 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/addi.S -lc -o rv64ui-v-addi
riscv64-unknown-elf-gcc  -DENTROPY=19165 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/add.S -lc -o rv64ui-v-add
Makefile:77: recipe for target 'rv64ui-v-simple' failed
make[1]: *** [rv64ui-v-simple] Error 1
riscv64-unknown-elf-gcc  -DENTROPY=19167 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/beq.S -lc -o rv64ui-v-beq
riscv64-unknown-elf-gcc: error: ./../env/v/entry.S: No such file or directory
riscv64-unknown-elf-gcc: error: ./../env/v/vm.c: No such file or directory
make[1]: Leaving directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
riscv64-unknown-elf-gcc: error: ./../env/v/entry.S: No such file or directory
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-simple.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-add.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-addi.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-and.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-andi.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-auipc.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-beq.hex
make -C /home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/ rv64ui-v-bge.hex
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
make[1]: Entering directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
riscv64-unknown-elf-gcc  -DENTROPY=19162 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/simple.S -lc -o rv64ui-v-simple
[... 8 lines cut ...]
Makefile:77: recipe for target 'rv64ui-v-simple' failed
make[1]: *** [rv64ui-v-simple] Error 1
riscv64-unknown-elf-gcc  -DENTROPY=19167 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/beq.S -lc -o rv64ui-v-beq
riscv64-unknown-elf-gcc: error: ./../env/v/entry.S: No such file or directory
riscv64-unknown-elf-gcc: error: ./../env/v/vm.c: No such file or directory
make[1]: Leaving directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
riscv64-unknown-elf-gcc: error: ./../env/v/entry.S: No such file or directory
make[1]: Leaving directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
Makefile:77: recipe for target 'rv64ui-v-beq' failed
make[1]: *** [rv64ui-v-beq] Error 1
make[1]: Leaving directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
/home/dfong/workspace/riscv/rocket-chip/Makefrag:43: recipe for target '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-bge.hex' failed
make: *** [/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-bge.hex] Error 2
Makefile:77: recipe for target 'rv64ui-v-addi' failed
make[1]: *** [rv64ui-v-addi] Error 1
make[1]: Leaving directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
/home/dfong/workspace/riscv/rocket-chip/Makefrag:43: recipe for target '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-and.hex' failed
make: *** [/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-and.hex] Error 2
make[1]: Leaving directory '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa'
/home/dfong/workspace/riscv/rocket-chip/Makefrag:43: recipe for target '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-add.hex' failed
make: *** [/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-add.hex] Error 2
/home/dfong/workspace/riscv/rocket-chip/Makefrag:43: recipe for target '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-beq.hex' failed
make: *** [/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-beq.hex] Error 2
/home/dfong/workspace/riscv/rocket-chip/Makefrag:43: recipe for target '/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-addi.hex' failed
make: *** [/home/dfong/workspace/riscv/rocket-chip/riscv-tools/riscv-tests/isa/rv64ui-v-addi.hex] Error 2

sbeamer, 2016-02-22:
`env` should be "installed" as part of pulling a git submodule. Please enter the 5 commands from the [Rocket Chip README.](https://github.com/ucb-bar/rocket-chip#checkout-the-code) It seems like you might have forgotten the last one.

davidfongsj1108, 2016-02-22:
Thanks for your quick help.

I think this step was very long
git submodule update --init
and I forgot to do this step
git submodule update --init --recursive riscv-tests

After I ran the above step and then
make -j8 run-asm-tests
how do I know if the test ran properly (so that I can have confidence in the installation flow)?
I can view the output/*.out but there is no indication of PASS/FAIL.

ccelio, 2016-02-22:
No indication is a sign of success (or more precisely, the return code of the execution command given to the Makefile marks a success).  Failures and assertions are printed to the *.out files.

closed 2016-02-22 by ccelio