chipsalliance/rocket-chip #53: Where to download the untethered lowRisc by Wei Song ?

davidfongsj1108, 2016-02-22:
Hi,

I'm a little confused by the various implementations of RISCV :

RocketChip (original SOC from UCB with HTIF)
zscale (RISCV 3 pipeline with no cache)

but where to find the untethered lowRISC by Wei Song ?

Or is the untethered lowRISC now the official rocket-chip release?
If yes,  how to add the "minion" RISCV to the rocket-chip?

Thanks,

David

ccelio, 2016-02-22:
lowrisc.org, to get the lowRISC fork. 

lowRISC is a fork, it is not "official" rocket-chip.  Occasionally Wei will push back fixes to us.  Rocket-chip is working on its own untethered implementation, which will hopefully not be too divergent from lowRISC.

closed 2016-02-22 by ccelio

davidfongsj1108, 2016-02-23:
The links I see on lowrisc.org points to the rocket-chip on github
https://github.com/ucb-bar/rocket-chip.git

Where is the one for lowrisc.org?

ccelio, 2016-02-23:
You should contact them about needing to add a "Download" link that points to their implementation. It's not obvious how to find their code from their webpage.

Under their community tab, they point to their github account:

https://github.com/lowrisc

davidfongsj1108, 2016-02-23:
Yeah. It's not obvious on how to download.
Thanks a lot !!