chipsalliance/rocket-chip #57: Which file allows one to change ICACHE, DCACHE sizes and number of RocketChip cores?

davidfongsj1108, 2016-02-23:
Hi,

Which configuration file allows one to change ICACHE, DCACHE sizes and number of RocketChip cores?

Thanks,

David

colinschmidt, 2016-02-23:
src/main/scala/Configs.scala contains the configuration for rocket-chip

There are groups of parameters for the L1 instruction and data caches named L1I and L1D. Be aware that not all combinations of values have been tested or are even valid.
NTiles controls the number of cores generated.

The values we think are reasonable to change given the current design are identified as "Knob"s and are all collected in the "knobValues" group on line 246.

davidfongsj1108, 2016-02-23:
I think it's in
src/main/scala/Configs.scala
  knobValues = {
    case "NTILES" => 1
    case "NBANKS_PER_MEM_CHANNEL" => 1
    case "L1D_MSHRS" => 2
    case "L1D_SETS" => 64
    case "L1D_WAYS" => 4
    case "L1I_SETS" => 64
    case "L1I_WAYS" => 4
    case "L1I_BUFFER_WAYS" => false
  }

What does L1D_MSHRS mean ?
I'm thinking of setting very small sizes and 2 cores like below:

  knobValues = {
    case "NTILES" => 2
    case "NBANKS_PER_MEM_CHANNEL" => 1
    case "L1D_MSHRS" => 1
    case "L1D_SETS" => 4
    case "L1D_WAYS" => 1
    case "L1I_SETS" => 4
    case "L1I_WAYS" => 1
    case "L1I_BUFFER_WAYS" => false
  }

colinschmidt, 2016-02-23:
There is a "SmallConfig" in that file that sets other parameters to make the core small, for example disabling the PFU. 

L1D_MSHRS is the number of Miss Status Handling Registers for the data cache

davidfongsj1108, 2016-02-23:
Thanks for the tip.
Do I modify this line in the "rocket-chip/vsim/Makefile"

CONFIG ?= DefaultVLSIConfig

 to 

CONFIG ?= SmallConfig

so that the settings in SmallConfig get called?

colinschmidt, 2016-02-23:
You can do that if you want, but I usually use the environment variable rather than modify the Makefile i.e.

`rocket-chip/vsim$ make CONFIG=SmallConfig`

or

`rocket-chip/vsim$ export CONFIG=SmallConfig`
`rocket-chip/vsim$ make`

davidfongsj1108, 2016-02-24:
Thanks for the tip.
You can close this issue.

closed 2016-02-24 by ccelio