chipsalliance/rocket-chip #59: Error with "make verilog CONFIG=SmallConfig"

davidfongsj1108, 2016-02-24:
Hi,

I'm getting an error using the cmd-line below (see (1) below)

cd rocket-chip/vsim
make verilog CONFIG=SmallConfig

Is there something wrong in my cmd-line setting and what's the correction?
I didn't modify any other files like Makefile

Thanks,

David

(1) screen output from cmd-line above is below:

cd /home/dfong/workspace/riscv/rocket-chip && mkdir -p /home/dfong/workspace/riscv/rocket-chip/vsim/generated-src && java -Xmx2048M -Xss8M -XX:MaxPermSize=256M -jar sbt-launch.jar "run rocketchip Top SmallConfig --W0W --minimumCompatibility 3.0.0 --backend rocketchip.RocketChipBackend --configName SmallConfig --compileInitializationUnoptimized --targetDir /home/dfong/workspace/riscv/rocket-chip/vsim/generated-src --configDump --noInlineMem"
[info] Loading project definition from /home/dfong/workspace/riscv/rocket-chip/project
Using addons: 
[info] Set current project to rocketchip (in build file:/home/dfong/workspace/riscv/rocket-chip/)
[info] Running rocketchip.TestGenerator rocketchip Top SmallConfig --W0W --minimumCompatibility 3.0.0 --backend rocketchip.RocketChipBackend --configName SmallConfig --compileInitializationUnoptimized --targetDir /home/dfong/workspace/riscv/rocket-chip/vsim/generated-src --configDump --noInlineMem
**[error] (run-main-0) java.lang.reflect.InvocationTargetException**
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
	at rocketchip.TestGenerator$$anonfun$15.apply(Testing.scala:148)
	at rocketchip.TestGenerator$$anonfun$15.apply(Testing.scala:145)
	at Chisel.Driver$$anonfun$executeUnwrapped$2$$anonfun$apply$1.apply(Driver.scala:96)
	at Chisel.Driver$$anonfun$executeUnwrapped$2$$anonfun$apply$1.apply(Driver.scala:96)
	at Chisel.Module$.Chisel$Module$$init(Module.scala:65)
	at Chisel.Module$.apply(Module.scala:50)
	at Chisel.Driver$$anonfun$executeUnwrapped$2.apply(Driver.scala:96)
	at Chisel.Driver$$anonfun$executeUnwrapped$2.apply(Driver.scala:96)
	at Chisel.Driver$.execute(Driver.scala:101)
	at Chisel.Driver$.executeUnwrapped(Driver.scala:96)
Caused by: cde.ParameterUndefinedException: Parameter HtifKey undefined.
	at cde.World$TopLookup$1.apply(Parameters.scala:150)
	at cde.Parameters.apply(Parameters.scala:313)
	at rocketchip.HasTopLevelParameters$class.htifW(RocketChip.scala:45)
	at rocketchip.BasicTopIO.htifW$lzycompute(RocketChip.scala:70)
	at rocketchip.BasicTopIO.htifW(RocketChip.scala:70)
	at rocketchip.BasicTopIO.<init>(RocketChip.scala:72)
	at rocketchip.TopIO.<init>(RocketChip.scala:76)
	at rocketchip.Top.<init>(RocketChip.scala:96)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
	at rocketchip.TestGenerator$$anonfun$15.apply(Testing.scala:148)
	at rocketchip.TestGenerator$$anonfun$15.apply(Testing.scala:145)
Caused by: scala.MatchError: HtifKey (of class uncore.HtifKey$)
	at rocketchip.SmallConfig$$anonfun$21.apply(Configs.scala:352)
	at rocketchip.SmallConfig$$anonfun$21.apply(Configs.scala:352)
	at cde.World$TopLookup$1.apply(Parameters.scala:148)
	at cde.Parameters.apply(Parameters.scala:313)
	at rocketchip.HasTopLevelParameters$class.htifW(RocketChip.scala:45)
	at rocketchip.BasicTopIO.htifW$lzycompute(RocketChip.scala:70)
	at rocketchip.BasicTopIO.htifW(RocketChip.scala:70)
	at rocketchip.BasicTopIO.<init>(RocketChip.scala:72)
	at rocketchip.TopIO.<init>(RocketChip.scala:76)
	at rocketchip.Top.<init>(RocketChip.scala:96)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:1943)
[... 59 lines cut ...]
	at rocketchip.TopIO.<init>(RocketChip.scala:76)
	at rocketchip.Top.<init>(RocketChip.scala:96)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:1943)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:1943)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:1943)
	at sbt.Defaults$.toError(Defaults.scala:38)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:719)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:717)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:run) Nonzero exit code: 1
[error] Total time: 6 s, completed Feb 23, 2016 4:57:58 PM
/home/dfong/workspace/riscv/rocket-chip/vsim/Makefrag-verilog:6: recipe for target '/home/dfong/workspace/riscv/rocket-chip/vsim/generated-src/Top.SmallConfig.v' failed
make: *** [/home/dfong/workspace/riscv/rocket-chip/vsim/generated-src/Top.SmallConfig.v] Error 1

jwright6323, 2016-02-24:
It looks like your setup can't find the HtifKey which might mean you haven't populated the uncore submodule.

Can you try going to the root of your git repo and running "git submodule update --init --recursive" and then retry running make?

zhemao, 2016-02-24:
The SmallConfig isn't meant to be used directly. Try using ExampleSmallConfig. Sorry for the confusion.

closed 2016-02-24 by zhemao

davidfongsj1108, 2016-02-24:
Thanks for the feedback.

Yes. when I used this line

make verilog CONFIG=ExampleSmallConfig

It runs through with no errors