chipsalliance/rocket-chip #60: How to create a configuration with ExampleSmallConfig and 2 CPU Cores and build synthesizeable verilog?

davidfongsj1108, 2016-02-24:
Hi,

If I want to create a configuration with ExampleSmallConfig and 2 CPU Cores (NTiles=2)
and create the synthesizable verilog,
what do I need to do?

Thanks,

David

zhemao, 2016-02-24:
You can create a new configuration combining SmallConfig and With2Cores. Add this to your Configs.scala

    class DualCoreSmallConfig extends Config(new With2Cores ++ new SmallConfig ++ new DefaultConfig)

davidfongsj1108, 2016-02-24:
I updated the Configs.scala and put the new configuration after ExampleSmallConfig
but I'm getting errors below which is baffling since there are other DefaultConfig.
Any ideas on what's wrong?

Configs.scala (small snippet)

class ExampleSmallConfig extends Config(new SmallConfig ++ new DefaultConfig)

class DualCoreSmallConfig extends Config(new With2Cores ++ new SmallConfig ++ new DefaultConfig)

make verilog CONFIG=DualCoreSmallConfig

cd /home/dfong/workspace/riscv/rocket-chip && mkdir -p /home/dfong/workspace/riscv/rocket-chip/vsim/generated-src && java -Xmx2048M -Xss8M -XX:MaxPermSize=256M -jar sbt-launch.jar "run rocketchip Top DualCoreSmallConfig --W0W --minimumCompatibility 3.0.0 --backend rocketchip.RocketChipBackend --configName DualCoreSmallConfig --compileInitializationUnoptimized --targetDir /home/dfong/workspace/riscv/rocket-chip/vsim/generated-src --configDump --noInlineMem"
[info] Loading project definition from /home/dfong/workspace/riscv/rocket-chip/project
Using addons: 
[info] Set current project to rocketchip (in build file:/home/dfong/workspace/riscv/rocket-chip/)
[info] Compiling 2 Scala sources to /home/dfong/workspace/riscv/rocket-chip/target/scala-2.11/classes...
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:16: DefaultConfig is already defined as class DefaultConfig
[error] class DefaultConfig extends Config (
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:255: DefaultVLSIConfig is already defined as class DefaultVLSIConfig
[error] class DefaultVLSIConfig extends DefaultConfig
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:256: DefaultCPPConfig is already defined as class DefaultCPPConfig
[error] class DefaultCPPConfig extends DefaultConfig
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:258: With2Cores is already defined as class With2Cores
[error] class With2Cores extends Config(knobValues = { case "NTILES" => 2 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:259: With4Cores is already defined as class With4Cores
[error] class With4Cores extends Config(knobValues = { case "NTILES" => 4 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:260: With8Cores is already defined as class With8Cores
[error] class With8Cores extends Config(knobValues = { case "NTILES" => 8 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:262: With2BanksPerMemChannel is already defined as class With2BanksPerMemChannel
[error] class With2BanksPerMemChannel extends Config(knobValues = { case "NBANKS_PER_MEM_CHANNEL" => 2 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:263: With4BanksPerMemChannel is already defined as class With4BanksPerMemChannel
[error] class With4BanksPerMemChannel extends Config(knobValues = { case "NBANKS_PER_MEM_CHANNEL" => 4 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:264: With8BanksPerMemChannel is already defined as class With8BanksPerMemChannel
[error] class With8BanksPerMemChannel extends Config(knobValues = { case "NBANKS_PER_MEM_CHANNEL" => 8 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:266: With2MemoryChannels is already defined as class With2MemoryChannels
[error] class With2MemoryChannels extends Config(
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:271: With4MemoryChannels is already defined as class With4MemoryChannels
[error] class With4MemoryChannels extends Config(
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:277: WithL2Cache is already defined as class WithL2Cache
[error] class WithL2Cache extends Config(
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:303: WithPLRU is already defined as class WithPLRU
[error] class WithPLRU extends Config(
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:308: WithL2Capacity2048 is already defined as class WithL2Capacity2048
[error] class WithL2Capacity2048 extends Config(knobValues = { case "L2_CAPACITY_IN_KB" => 2048 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:309: WithL2Capacity1024 is already defined as class WithL2Capacity1024
[error] class WithL2Capacity1024 extends Config(knobValues = { case "L2_CAPACITY_IN_KB" => 1024 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:310: WithL2Capacity512 is already defined as class WithL2Capacity512
[error] class WithL2Capacity512 extends Config(knobValues = { case "L2_CAPACITY_IN_KB" => 512 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:311: WithL2Capacity256 is already defined as class WithL2Capacity256
[error] class WithL2Capacity256 extends Config(knobValues = { case "L2_CAPACITY_IN_KB" => 256 })
[error]       ^
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:312: WithL2Capacity128 is already defined as class WithL2Capacity128
[error] class WithL2Capacity128 extends Config(knobValues = { case "L2_CAPACITY_IN_KB" => 128 })
[... 241 lines cut ...]
[... cut ...]
[error] class QuadChannelBenchmarkConfig extends Config(new With4MemoryChannels ++ new SingleChannelBenchmarkConfig)
[error]       ^
[error] 51 errors found
[error] (rocketchip/compile:compileIncremental) Compilation failed
[error] Total time: 8 s, completed Feb 24, 2016 11:38:02 AM
/home/dfong/workspace/riscv/rocket-chip/vsim/Makefrag-verilog:6: recipe for target '/home/dfong/workspace/riscv/rocket-chip/vsim/generated-src/Top.DualCoreSmallConfig.v' failed
make: *** [/home/dfong/workspace/riscv/rocket-chip/vsim/generated-src/Top.DualCoreSmallConfig.v] Error 1

davidfongsj1108, 2016-02-25:
Need help with how to reinitialize the build/make flow under

vsim

Looks like there is something counting the number of Scala sources "**Compiling 2 Scala sources**"

I think this is causing the compilation problem.

info] Set current project to rocketchip (in build file:/home/dfong/workspace/riscv/rocket-chip/)
[info] **Compiling 2 Scala sources **to /home/dfong/workspace/riscv/rocket-chip/target/scala-2.11/classes...
[error] /home/dfong/workspace/riscv/rocket-chip/src/main/scala/Configs.scala:16: DefaultConfig is already defined as class DefaultConfig

davidfongsj1108, 2016-02-25:
I think problem is that I used this cmd-line

make verilog CONFIG=ExampleSmallConfig |& tee make.log

Somehow the pipe "|" and "tee" causes problem in the make process.

When I installed a new build and didn't add the " | & tee make.log" then
the make runs with no more errors.

davidfongsj1108, 2016-02-25:
Is there a safe way to log the stdout to a file?

Also, when I updated the DualCoreSmallConfig in the Configs.scala,
make verilog CONFIG=DualCoreSmallConfig
it ran with no errors.

zhemao, 2016-02-26:
Couldn't you do

    make verilog CONFIG=ExampleSmallConfig 2>&1 | tee make.log

davidfongsj1108, 2016-03-02:
I was able to run the "make verilog CONFIG=ExampleSmallConfig" and no strange installation or error occurs.  Please close this issue.

closed 2016-03-02 by sbeamer