chipsalliance/rocket-chip #61: Where to find systemverilog code for AXI slave with DDR memory controller?

davidfongsj1108, 2016-02-25:
Hi,

I was able to generate the verilog for the rocketchip.

However, I saw there is an AXI like interface "io_mem_*".
Is there systemverilog code that takes the AXI -like master signals from Top
and drives an AXI slave with DDR memory controller?

Thanks,

David

zhemao, 2016-02-25:
For our FPGA flow, we use the AXI DDR memory controllers provided by Xilinx. For our taped out chips, we use the backup memory.

If you are using a different FPGA architecture or silicon technology, you'll have to find your own memory controller.

davidfongsj1108, 2016-02-25:
Please explain "backup memory" for the taped out chips.

colinschmidt, 2016-02-25:
"backup memory" is a slow digital IO that uses a non-standard (ucb-bar specific) format

davidfongsj1108, 2016-03-02:
Ok. Please close this issue.

closed 2016-03-02 by aswaterman