chipsalliance/rocket-chip #64: Device tree generation

wsong83, 2016-02-25:
I see there is an automatic device tree generation in the current Rocket-chip generator.
Can I somehow get a dts file from it instead a dtb file?
I am trying to catch up with the latest development.
Is the generated dtb file utilized in emulator/vsim/FPGA?
Also is the current RISC-V Linux kernel reads this file?

Thanks!

zhemao, 2016-03-01:
The generated dtb file is put into a ROM. Our intention is that the machine-mode software will read the dtb ROM to gather data about what peripherals are available. There shouldn't be any need to ship .dtb or .dts files along with the kernel.

wsong83, 2016-03-02:
I see. Thanks for the clarification.
Just think an automatic generated dts might help debug but I can read the Chisel device tree to figure out.

aswaterman, 2016-03-02:
It will be a moot point when we release the platform spec, as the new
config string is human readable so can just be printed to console. In other
words, I agree it is helpful for debugging ;-)

palmer-dabbelt, 2016-03-22:
@aswaterman I see your not-device-tree thing getting printed in my Rocket Chip builds, so I'm closing this.

closed 2016-03-22 by palmer-dabbelt