chipsalliance/rocket-chip #74: Atomic increment using LR/SC

mn416, 2016-03-15:
Hi,

I have a barrier implementation which works when the atomic increment part is implemented using an `amoadd.d`:

```
amoadd.d a0, %1, (%0)
```

but not when using an LR/SC:

```
1:
lr.d   a0, 0(%0)
add    a0, a0, %1
sc.d   a0, a0, 0(%0)
bnez   a0, 1b
```

I've made a simplified, self-contained implementation that demonstrates the problem:

1. Download the source files at:

  http://www.cl.cam.ac.uk/~mn416/rocket-lrsc-issue/

2. Build by running `make.sh` to produce `main.hex`

3. Pass `main.hex` to a rocket-chip emulator compiled with `NTILES=2':

```
> ./emulator +loadmem=main.hex
```

When using `amoadd.d`, the program prints

```
Started
Finished
```

When using LR/SC, we only see `Started`.  The second half of the barrier, which waits until every core has incremented a shared variable, doesn't seem to terminate.

colinschmidt, 2016-03-15:
Does the same test work in spike with -p2 ?

mn416, 2016-03-15:
Yes, with the LR/SC code, `spike -p2 main.elf` gives

```
Started
Finished
```

as expected.

aswaterman, 2016-03-15:
I am looking into this.

aswaterman, 2016-03-15:
This bug seems to date back to 8b5a2e964b99030eeb44d34396915d8af88b3727, when TileLink Finish messages were removed from the L1 D$.

@hcook the L1 was waiting to send finish messages until the replay queue had drained, which prevented a race between replays and subsequent probes.  In this case, it manifests as livelock on LR/SC, because the probe occurs before the LR replays, so the SC is never able to complete.  I think the cleanest option is to use finish messages.  It's not obvious to me how else to solve this without blocking all probes to addresses that have outstanding Acquires, which would seem to cause deadlock.  (It doesn't suffice to block probes between when the grant occurs and when the replay queue is drained, because the probe could actually arrive before the grant.)

commit bd3dba7f (2016-04-02): Fix LR/SC livelock bug

Closes #74.
closed 2016-04-02 by hcook