chipsalliance/rocket-chip #86: Atomic swap using LR/SC

mn416, 2016-04-05:
The `tracegen+check.sh` script is showing some odd LR/SC behaviour:

  1. a very low LR/SC success rate;
  2. successful SCs that are not atomic.

I'm still trying to figure out the reason for (1), but I've managed to reproduce (2) in an ISA-level test.  I will focus on this ISA-level test below.  (If you'd prefer to see it using `tracegen+check.sh` then you need to uncomment line 429 in `tracegen.scala` and comment line 430.)

The problem seems to occur when there is no dependency between an LR and an SC, as would arise when implementing an atomic swap using LR/SC.  In other words: when the SC is issued before the LR response is received.

I'm using litmus tests to observe the bad behaviour.  (I ported [CHERI-Litmus](https://github.com/CTSRD-CHERI/CHERI-Litmus) to rocket-chip.)

Here is the litmus test:

```
RISCV
"LR/SC atomic swap test"
{
0:r2=x;
1:r2=x;
}
 P0                  | P1 ;
 lr.w r1, 0(r2)      | lr.w r1, 0(r2) ;
 li r3, 1            | li r3, 2 ; 
 sc.w r4, r3, 0(r2)  | sc.w r4, r3, 0(r2) ;
exists
(0:r4=0 /\ 1:r4=0 /\ 0:r1=0 /\ 1:r1=0)
```

Unpacking this:

  * we have one shared variable `x`, implicitly initialised to 0
  * we have two processes, both performing an LR/SC to `x`
  * one process writes `1`, the other writes `2`
  * the `exists` condition is looking for the case where both SCs succeed and both LRs return 0, which should be impossible

Running the test we get:

```
Observed outcomes:
751: 0:r4=0 1:r4=0 0:r1=0 1:r1=1
1: 0:r4=1 1:r4=0 0:r1=0 1:r1=0
248: 0:r4=0 1:r4=0 0:r1=0 1:r1=0
OBSERVED
```

The condition that should be impossible is observed 248 times out of a 1000.

When I replace the two `li` instructions with `add r3, r1, 1`, i.e. turning the swaps into increments and introducing a dependency between the LR and SC, then I am unable to observe any bad behaviour.

Elfs, objdumps, and hexes are available here:

http://www.cl.cam.ac.uk/~mn416/lrsc-issue/

The test takes about a minute to complete.

aswaterman, 2016-04-05:
I observe the same outcome... I'll look into it.

aswaterman, 2016-04-05:
@hcook this is because of a control bug in the L1 MSHR causing the line not to be marked dirty when it should.

The LR is accepted as the primary miss.  Then the SC is accepted as a secondary miss, but new_coh_state isn't updated to indicate the line will become dirty.

The erroneous assumption is that secondary misses should only change the ultimate coh_state if cmd_requires_second_acquire is true.  This breaks down for LR/SC because LR gets the line exclusive, so SC doesn't need a second acquire -- but LR didn't mark the line dirty, whereas SC should.

One easy fix is to tighten requiresAcquireOnSecondaryMiss to

    isWriteIntent(second_cmd) && !isWrite(first_cmd)

and so, while LR will get the line exclusive, the MSHR will treat the SC as needing an upgrade.  This will basically never affect performance, because there's usually a RAW hazard between LR and SC.

hcook, 2016-04-06:
That seems like a reasonable fix, but it seems wasteful to block Stores missing under PF.Ws, no? I can look into whether we can actually get the right coherence state as we drain the replay queue.

aswaterman, 2016-04-06:
True.

I noticed that new_coh_state can be updated from two places when in the
refill_resp state. At the moment they're mutually exclusive because of the
needs-second-acquire thing, but that need some attention otherwise.

mn416, 2016-04-12:
I had another go at figuring out the low LR/SC success rate.

Looks like an issue in groundtest.  Adding the following line to `GroundTestTile` in `tile.scala` seems to solve the problem:

```
dcache.io.cpu.invalidate_lr := Bool(false)
```

If this sounds sensible @zhemao @hcook @aswaterman, I can commit a patch.

aswaterman, 2016-04-20:
@mn416 please do.

@hcook any further thoughts on the MSHR control?

zhemao, 2016-05-15:
I think we forgot about this issue. Did we ever get that groundtest patch in or resolve the atomicity bug?

mn416, 2016-05-16:
Hi @zhemao, I get back to work at the end of the week and will submit a groundtest patch then.  I notice that the trace generator is currently giving "DCache exception occurred - cache response not killed", so I'll also look into this regression.

zhemao, 2016-07-22:
I think this has already been fixed by [this commit](https://github.com/ucb-bar/groundtest/commit/8145f3ed05328077be6c823752dea0325e367ac5). Matt is that correct? If so, I will close this issue.

mn416, 2016-07-22:
Hi @zhemao, that commit resolves the low success rate but the atomicity bug is still observable.

In `TraceGen.scala` we can see these three lines:

```
      // Delay finished, send SC request
      // when (opInProgress === UInt(3)) {
      when (opInProgress === UInt(3) && !reqValid && reqCount === respCount) {
```

Here, an LR has just been sent.  The third line waits until all outstanding requests have been responded to, before sending an SC.  While this hides the bug, it shouldn't be necessary: the rocket pipeline is capable of submitting an SC before the LR response comes back (that's what the litmus test told us), so we should test this behaviour in TraceGen.

So, commenting the third line and uncommenting the second we get:

```
> tracegen+check.sh
Testing against WMO model:
 
       0: 

Failed WMO with seed 0
See tracegen-log/trace.txt and tracegen-log/trace.axe for counterexample
```

To see the problem, we can do:

```
> axe-shrink.py wmo tracegen-log/trace.axe
Pass 0
Omitted 281 of 289         
Pass 1
Omitted 286 of 289         
Pass 2
Omitted 286 of 289         
1: M[5] := 29 @ 743:
1: { M[5] == 29; M[5] := 53} @ 1047:1088
0: { M[5] == 29; M[5] := 44} @ 1058:1126
```

The { } syntax denotes an atomic RMW -- an LR/SC pair where the SC succeeded. If either of the RMWs were actually atomic, then it should be impossible for them to both read the same value.

zhemao, 2016-07-26:
I'm taking a look at this now. Where in the tracegen code do you check that the store-conditional was successful?

mn416, 2016-07-26:
Hi @zhemao, tracegen doesn't actually look to see if SC is successful or not.  It just emits the response from the SC regardless, which (from what I remember) is zero for success and non-zero for failure.

When the trace is converted to axe format, then the SC response is examined.  On success, the LR/SC pair is turned into an atomic RMW using axe's "{}" syntax.  On failure, the SC is dropped and the LR is treated like a standard load.

zhemao, 2016-07-27:
Alright, it's fixed. As Andrew mentioned, the problem is that the MSHR fails to upgrade the state from clientExclusiveClean to clientExclusiveDirty if the SC comes in as a secondary request. The solution is to track whether secondary requests will dirty the line and then change the state accordingly.

closed 2016-07-27 by zhemao