chipsalliance/rocket-chip #89: Chisel3 generation creates empty begin/end bodies

mwachs5, 2016-04-15:
From vsim, when I build the verilog with CHISEL_VERSION=3 CONFIG=DefaultRV32Config, there are many empty begin-end bodies generated. This is legal in SystemVerilog, but not Verilog, and some tools complain.

Is this sort of output expected? I can set my tool to accept SystemVerilog inputs but this could cause other issues.

Example: first occurrence is in HTIF, line 613 of my generated output:

  always @(posedge clk) begin
    if(reset) begin
      rx_count <= 15'h0;
    end else begin
      if(T_1391) begin
        if(T_1392) begin
          rx_count <= 1'h0;
        end else begin
          if(T_1225) begin
            rx_count <= T_1228;
       **   end else begin
            ;
          end**
        end

...

mwachs5, 2016-04-15:
To clarify, it's specifically the semicolon which is making the tool I am using unhappy. Without the semicolon, the empty body is OK.

aswaterman, 2016-04-15:
For the chisel3 flow, Verilog code generation issues should be raised against FIRRTL (https://github.com/ucb-bar/firrtl) -- I'll take care of it, though.

closed 2016-04-20 by aswaterman