chipsalliance/rocket-chip #97: emulator make error

jia1546, 2016-04-28:
I want to build a C emulator that is capable of generating VCD waveform, so I enter the directory of emulator and compile with "make debug". Then I find the following errors:
[error_info.txt](https://github.com/ucb-bar/rocket-chip/files/239651/error_info.txt)

aswaterman, 2016-04-28:
Looks like gcc is dying on you.

Are you on a machine with limited RAM?  What version of g++ are you using?

jia1546, 2016-04-28:
My gcc version is 5.3.1 20160413

jia1546, 2016-05-04:
I am working on VM, and I set the memory to 16G and use swap but the above error still exists.@aswaterman

aswaterman, 2016-05-04:
It's possible that Chisel debug mode with 2 rocket cores is just too much
for GCC to handle. You can try using -O0 instead of -O1 -- that usually
reduces the compiler's memory footprint and runtime substantially.

jia1546, 2016-05-05:
After I reduce the number of tiles to 1, the build of emulator succeed.  And this confirms your point that the above error was caused by a limited RAM. Thanks! @aswaterman

zhemao, 2016-05-15:
One way to avoid having the emulator build crap out due to limited RAM is to reduce how many g++ jobs run at the same time. In [emulator/Makefile line 29](https://github.com/ucb-bar/rocket-chip/blob/master/emulator/Makefile#L29), change "-j" to something like "-j N" to run with only N threads. Or just get rid of it entirely to run single-threaded.

littleb0y33, 2016-05-24:
I got the same problem, but with g++ 4.8.4 and 4Gb of RAM + 2Gb Swap on VM Linux 14.04.

@zhemao I tried to run it single-threaded and the same problem happens. 

How do I reduce that number of tiles to make it run ?

Ssavas, 2016-05-30:
Hi,

Number of tiles are represented as 'NTiles' in Configs.scala (src/main/scala/Configs.scala) however I cannot find where it is set. It would be nice to know.

jia1546, 2016-05-31:
in Configs.scala, line 296 says case NTiles => Knob("NTILES") and line 310 says case "NTILES" => 1. So, the above configuration sets "NTiles" to 1 at last.

littleb0y33, 2016-05-31:
I managed to build the emulator with 8Gb of RAM, single threaded and simulating one core. I didn't find how to reduce the number of cores, so I used aswaterman's solution. 
But somehow I can't explain why it is using so much RAM for a simple chip like rocket.

closed 2016-07-21 by hcook