chipsalliance/rocket-chip #104: DivSqrtRecF64 is no longer retime-able in synthesis

ben-k, 2016-05-11:
The divide/square-root unit in the FPU used to be a valid target for retiming, but recently I've been getting the following message from the synthesis tool running on DefaultVLSIConfig:

```
Information: Checking pipeline property of cell RocketTile/FPU/DivSqrtRecF64 (design DivSqrtRecF64). 
Information: Pipeline detection aborted. Reason: cell RocketTile/FPU/DivSqrtRecF64/ds/*cell*151366 is in a feedback loop.
Information: Aborted pipeline detection on cell RocketTile/FPU/DivSqrtRecF64 (design DivSqrtRecF64).
```

Is this no longer a valid pipeline for retiming?  Do we care?

aswaterman, 2016-05-11:
I know @yunsup spent a decent amount of time playing around with it, so maybe he can chime in.

The unit does have a feedback loop.  There's a three-stage mul-add that feeds back into itself; but the mul-add is written as (a*b+c) -> reg -> reg, so it is meant to be retimed.

colinschmidt, 2017-02-28:
If you specify the multiplier(Mul54) individually it is detected as a pipeline (this was the critical path on our chip). I'm still looking into this from a QoR perspective though, and will close/update at the end of my investigation.

terpstra, 2017-03-24:
@aswaterman So this issue is already resolved, right?

terpstra, 2017-05-17:
Pretty sure this has been fixed for a while now.

closed 2017-05-17 by terpstra