chipsalliance/rocket-chip #111: ExampleSmallConfig Fails run-bmark-tests

colinschmidt, 2016-05-19:
I believe the issue has to do with the fact that ExampleSmallConfig has no FPU. 
The code jumps to the trap_handler during the boot code, because it seems to execute the fssr rather than taking the branch around the FPU setup code.

At first I thought it might be due to a branch mis-predict but I think rather the check here

```
  ## if that didn't stick, we don't have a FPU, so don't initialize it                                                                                                              
  li t1, MSTATUS_FS                                                                                                                                                                 
  and t1, t0, t1                                                                                                                                                                    
  beqz t1, 1f   
```

is not actually branch around the FPU setup code.

Should be replicable on master with
```
make CONFIG=ExampleSmallConfig output/median.riscv.out
```

colinschmidt, 2016-05-19:
BTW I replicated this on spike without an FPU so its almost certainly the crt.S in riscv-tests. Let me know if its best to move this issue there.

```
spike --isa=RV64IMA -d median.riscv
```

colinschmidt, 2016-05-31:
ExampleSmallConfig now also fails assembly tests on rocket-chip/master

aswaterman, 2016-05-31:
This is known; the wrong set of tests are executed.  I'll fix this soon.

closed 2016-07-22 by zhemao