chipsalliance/rocket-chip #112: dataBits and dataBeats parameters in HasHtifParameters

jia1546, 2016-05-24:
```
trait HasHtifParameters {
  implicit val p: Parameters
  val htifExternal = p(HtifKey)
  val dataBits = p(TLKey(p(TLId))).dataBitsPerBeat
  val dataBeats = p(TLKey(p(TLId))).dataBeats
  val w = htifExternal.width
  val nSCR = htifExternal.nSCR
  val scrAddrBits = log2Up(nSCR)
  val scrDataBits = 64
  val scrDataBytes = scrDataBits / 8
  val csrDataBits = htifExternal.csrDataBits
  val csrDataBytes = csrDataBits / 8
  val offsetBits = htifExternal.offsetBits
  val nCores = htifExternal.nCores
}
```
In the above code snippet from htif.scala, the TLKey(p(TLId)) corresponds to TLKey("L1toL2") in Configs.scala. Thus, p(TLKey(p(TLId))) equals to the TileLinkParameters defined by TLKey("L1toL2"). However, neither dataBits or dataBeats shows in this definition.

xfguo, 2016-05-24:
@jia1546 
Check this file. https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/Configs.scala

jia1546, 2016-05-25:
yes, neither dataBits or dataBeats shows in the HasHtifParameters trait definition in Configs.scala. @xfguo

zhemao, 2016-05-26:
Those are from the TileLinkParameters object, not the HtifParameters object.

closed 2016-07-05 by zhemao