chipsalliance/rocket-chip #120: Heterogeneous Core configs do not generate correct L1toL2 network

colinschmidt, 2016-06-02:
The current L1toL2 network is parameterized assuming all Tiles have the same number of TileLink ports. (see the multiply in the following snippet from Configs.scala)

```
          nCachelessClients = site(ExtraL1Clients) +
                   site(NTiles) *
                    (1 + (if(site(BuildRoCC).isEmpty) 0
                           else site(RoccNMemChannels)))
```

Creating a config with only some tiles having a RoCC accelerator with an L2 port produces the bug.

colinschmidt, 2016-06-02:
@zhemao @hcook Thoughts on how to resolve this?

aswaterman, 2016-06-03:
This kind of thing doesn't belong in the parameters; it should be computed as the generator executes.  It will require a fair amount of refactoring.

zhemao, 2016-06-10:
Yeah, I agree with Andrew. I think if we want to do this, it should be done in the following way.

1. Create a new TLKey in the config, name it "TileL1". This gets changed per core.
2. In the OuterMemorySystem class, build up a new TLKey by adding up the number of channels for each core.
3. Use this new TLKey when instantiating the l1tol2net, coherence agent, and other TL modules at the L2 frontside.
4. Backside of the L2 can remain as is.

aswaterman, 2016-06-10:
That sounds like the right first step, but ultimately, the tile shouldn't have to precompute its own value; it should fall out of running the generator.

closed 2016-07-05 by zhemao