chipsalliance/rocket-chip #154: Infinite loop on rv64ui-v* tests using Chisel2

ccelio, 2016-07-07:
I'm cloning a clean copy of the rocket-chip repo, building the tools from scratch, and changing Chisel version in ./Makefrag to `CHISEL_VERSION ?= 2`.

Steps to reproduce:

````bash
git clone https://github.com/ucb-bar/rocket-chip.git clean-rocket-chip
cd clean-rocket-chip/
git submodule update --init
cd riscv-tools/
git submodule update --init --recursive
./build.sh
cd ..; vim Makefrag (change Chisel version to 2)
cd vsim/; make output/rv64ui-v-simple.out
... hangs ...
````

This only occurs in vsim.*

if I look at the processor printout, it's stuck on `sd gp, 24(sp)`:

````
C0:     103136 [1] pc=[ffffe000c8] W[r 2=ffffffffffe086d0][1] R[r 2=0000000000000000] R[r 0=0000000000000000] inst=[14011173] csrrw   sp, sscratch, sp
C0:     103137 [0] pc=[ffffe000c8] W[r 0=0000000000000000][0] R[r 2=0000000000000000] R[r 0=0000000000000000] inst=[14011173] csrrw   sp, sscratch, sp
C0:     103138 [0] pc=[ffffe000c8] W[r 0=0000000000000000][0] R[r 2=0000000000000000] R[r 0=0000000000000000] inst=[14011173] csrrw   sp, sscratch, sp
C0:     103139 [0] pc=[ffffe000cc] W[r 0=ffffffffffe086d8][0] R[r 2=ffffffffffe086d0] R[r 1=0000000000000000] inst=[00113423] sd      ra, 8(sp)
C0:     103140 [0] pc=[ffffe000d0] W[r 0=ffffffffffe086e8][0] R[r 2=ffffffffffe086d0] R[r 3=0000000000000000] inst=[00313c23] sd      gp, 24(sp)
C0:     103141 [0] pc=[ffffe000d0] W[r 0=ffffffffffe086e8][0] R[r 2=ffffffffffe086d0] R[r 3=0000000000000000] inst=[00313c23] sd      gp, 24(sp)
C0:     103142 [0] pc=[ffffe000d0] W[r 0=ffffffffffe086e8][0] R[r 2=ffffffffffe086d0] R[r 3=0000000000000000] inst=[00313c23] sd      gp, 24(sp)
C0:     103143 [0] pc=[ffffe000d0] W[r 0=ffffffffffe086e8][0] R[r 2=ffffffffffe086d0] R[r 3=0000000000000000] inst=[00313c23] sd      gp, 24(sp)
C0:     103144 [0] pc=[ffffe000d0] W[r 0=ffffffffffe086e8][0] R[r 2=ffffffffffe086d0] R[r 3=0000000000000000] inst=[00313c23] sd      gp, 24(sp)
... "sd gp, 24(sp)" ad infinitum ...
````

Is anybody else seeing this?  Is this a known problem?

--------

*Actually, I also hit an assertion failure in NASTI when trying "make run" in emulator:

````
 C0:      56299 [1] pc=[0080000304] W[r 2=0000000000ff00ff][1] R[r 3=0000000080002000] R[r 0=0000000000000000] inst=[0001a103] lw      sp, 0(gp)
 Invalid read address 701082c0                                                                                            
 C0:      56300 [0] pc=[0080000304] W[r 0=0000000080002000][0] R[r 3=0000000080002000] R[r 0=0000000000000000] inst=[0001a103] lw      sp, 0(gp)
 C0:      56301 [0] pc=[0080000304] W[r 0=0000000080002000][0] R[r 3=0000000080002000] R[r 0=0000000000000000] inst=[0001a103] lw      sp, 0(gp)
 Assertion failed: NASTI read error                                                                                       
 emulator-Top-DefaultConfig: ../fesvr/dtm.cc:297: void dtm_t::tick(bool, bool, dtm_t::resp): Assertion `resp_wait' failed.
````

zhemao, 2016-07-22:
I wonder if it's actually worth it to continue supporting Chisel 2. At some point, we'll want to start using Chisel3-specific features. Are you trying to do this in order to verify the Chisel3-ported BOOM against the Chisel2 version?

ccelio, 2016-07-22:
Yes, I really would appreciate a still-working chisel2 rocket-chip while porting BOOM; as I make a change to accommodate chisel3, I would like to verify I didn't break the functionality by running it in chisel2.  Instead, I just have to throw a hail mary and see what happens once I get to the end of it.

zhemao, 2016-07-27:
Which commit of rocketchip are you using? I think 19b44ec is the last commit still compatible with Chisel2. Have you tried running in the emulator?

ucbjrl, 2016-07-27:
Jenkins is able to build and test ecd1af326c49e8fbf27e32f8da0da1920c66ffcd with chisel2 (master TIP).

zhemao, 2016-07-27:
Well that's interesting. How does it get around the clock-crossing utilities in junctions, which are only compatible with chisel3?

ucbjrl, 2016-07-27:
I added code to Chisel2 so that they compile. I don't think they're actually used yet. It will probably stop working once they are used.

zhemao, 2016-07-27:
OK, thanks for that. I see your change in master. The clock-crossings aren't used in rocket chip right now, so it doesn't matter if they're non-functional in Chisel2. I'll try building the emulator again.

zhemao, 2016-07-27:
So I tried it with the latest RocketChip master plus Jim's fixes on chisel2 master and some uncore fixes.

The VSIM simulation does freeze, but the emulator seems to work just fine. I wonder if this has to do with the initial randomization of the registers.

zhemao, 2016-07-28:
Okay, so the problem was that the PTE cache in the PTW was given a non-power-of-two size, causing a read to an invalid entry in the PLRU state register. I've pushed a workaround in rocket that should make this work.

zhemao, 2016-07-28:
Currently, all the tests run to completion, but the fdiv and fcvt tests are failing for me. I'm not exactly sure why this is. But it seems to be working in the emulator, so maybe go with that for now? This probably has to do with some VCS X quirkiness.

ccelio, 2016-07-28:
I appreciate you looking into this.

closed 2016-09-07 by hcook