chipsalliance/rocket-chip #155: Synopsys-VCS emulator error

jia1546, 2016-07-08:
What is the usable version of Synopsys-vcs when emulating the rocket-chip? I used vcs-2014.03, but I got the following error when I typed "make" in the rocket-chip/vsim subdirectory.
```
_csrc0.so: undefined reference to `QRD_VCSgd_RecFNToIN_1_jIVhfe_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_3_TfB2eb_1'
_csrc0.so: undefined reference to `QRD_VCSgd_L2BroadcastHub_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_12_2yUUxc_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FPToInt_1'
_csrc0.so: undefined reference to `QRD_VCSgd_TileLinkEnqueuer_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ProbeUnit_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FinishQueue_0_9unCvd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BroadcastAeTracker_0_Jc4xyd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_0_Qyvlbb_1'
_csrc0.so: undefined reference to `QRD_VCSgd_IntToFP_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BroadcastAeTracker_1_2bZmDd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingRRArbiter_7_aV9ube_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BroadcastAeTracker_5_2J1lmd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ClientTilenkEnqueuer_eaOaic_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BasicCrossbar_2_kTNPed_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BroadcastAeTracker_3_GhDzsd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_2_6sS4Mc_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ROMSlave_0_hISk4b_1'
_csrc0.so: undefined reference to `QRD_VCSgd_RecFNToRecFN_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Frontend_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingArbiter_0_3XHOh_1'
_csrc0.so: undefined reference to `QRD_VCSgd_MulDiv_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_12_f3SeVb_1'
_csrc0.so: undefined reference to `QRD_VCSgd_MSHR_1_IeNvRb_1'
_csrc0.so: undefined reference to `QRD_VCSgd_MMIOTileLinkManager_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingRRArbiter_5_0gQH0d_1'
_csrc0.so: undefined reference to `QRD_VCSgd_MetadataArray_1'
_csrc0.so: undefined reference to `QRD_VCSgd_TileLinkIONarrower_1_xXLhbd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_HellaCacheArbiter_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FinishQueue_1_6atNqd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_5_FwQ3Oc_1'
_csrc0.so: undefined reference to `QRD_VCSgd_DataArray_1'
_csrc0.so: undefined reference to `QRD_VCSgd_SlowIO_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ClientUncaworkPort_0_ACkGE_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingRRArbiter_3_cANWPd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_DivSqrtRecF64_1'
_csrc0.so: undefined reference to `QRD_VCSgd_RocketCAM_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ICache_T178_I4Sv6b_1'
_csrc0.so: undefined reference to `QRD_VCSgd_PRCI_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ManagerTilworkPort_0_wbBDb_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ClientUncaworkPort_1_vqQmI_1'
_csrc0.so: undefined reference to `QRD_VCSgd_MSHRFile_1'
_csrc0.so: undefined reference to `QRD_VCSgd_TileLinkIONarrower_0_0vHT5c_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_4_IeSKP_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FPUFMAPipe_1_zat2Kd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ClientUncaOArbiter_2_uR9Uj_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ClientTileOUnwrapper_jtllye_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Top_1'
_csrc0.so: undefined reference to `QRD_VCSgd_AMOALU_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ClientTileetworkPort_uOE6L_1'
_csrc0.so: undefined reference to `QRD_VCSgd_PTW_1'
_csrc0.so: undefined reference to `QRD_VCSgd_HellaCache_1'
_csrc0.so: undefined reference to `QRD_VCSgd_WritebackUnit_1'
_csrc0.so: undefined reference to `QRD_VCSgd_SCRFile_1'
_csrc0.so: undefined reference to `QRD_VCSgd_OuterMemorySystem_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FPUFMAPipe_0_0JbMGd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_11_hwTXMc_1'
_csrc0.so: undefined reference to `QRD_VCSgd_RecFNToRecFN_1_3cSMgb_1'
_csrc0.so: undefined reference to `QRD_VCSgd_RTC_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BroadcastAeTracker_2_JZ7Sod_1'
_csrc0.so: undefined reference to `QRD_VCSgd_NastiIOTilonverter_1_zfOGRd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_5_dBoVK_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BroadcastAeTracker_6_fC289c_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FPToFP_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_4_YDXrUc_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ROMSlave_1_04wJ9b_1'
_csrc0.so: undefined reference to `QRD_VCSgd_NastiIOTilonverter_0_cLS4Wd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_1_Hs3v6_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_6_suWO1c_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_2_mKFJib_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BroadcastAeTracker_4_9l9whd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FPU_1'
_csrc0.so: undefined reference to `QRD_VCSgd_INToRecFN_1_7Y3Vmd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingRRArbiter_4_tqt65d_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BasicCrossbar_1_TA9Whd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ManagerTilworkPort_1_Vc5sg_1'
_csrc0.so: undefined reference to `QRD_VCSgd_TileLinkEnqueuer_1_XCfYV_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BasicCrossbar_0_AMRcmd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_rocketTestHarness_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BasicCrossbar_4_oSIdDd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_FPUDecoder_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ALU_1'
_csrc0.so: undefined reference to `QRD_VCSgd_TileLinkMeterconnect_xd7dhe_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_1_NYV2xc_1'
_csrc0.so: undefined reference to `QRD_VCSgd_IOMSHR_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingRRArbiter_6_FTrLfe_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BTB_1'
_csrc0.so: undefined reference to `QRD_VCSgd_TLB_1'
_csrc0.so: undefined reference to `QRD_VCSgd_TileLinkRerconnect_1_jQZqjd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_CSRFile_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingRRArbiter_1_8ojqHd_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Queue_6_MPIpZ_1'
_csrc0.so: undefined reference to `QRD_VCSgd_ReorderQueue_1_0xsDJb_1'
_csrc0.so: undefined reference to `QRD_VCSgd_BasicCrossbar_3_BmIr9c_1'
_csrc0.so: undefined reference to `QRD_VCSgd_RocketTile_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Arbiter_3_378OIc_1'
_csrc0.so: undefined reference to `QRD_VCSgd_Mul54_1'
_csrc0.so: undefined reference to `QRD_VCSgd_LockingRRArbiter_0_V4XOMd_1'
_csrc0.so: undefined reference to `M_VCSgd_rocketTestHarness_1'
```

The following file keeps the trace of the emulating process: 
[vcs-emulator-error.txt](https://github.com/ucb-bar/rocket-chip/files/353354/vcs-emulator-error.txt)

ccelio, 2016-07-21:
Building on IA-32 platforms is not supported.

closed 2016-07-21 by ccelio

kkiningh, 2017-03-31:
I had a similar problem. The relevant Solvnet article suggested that I had conflicting versions of VCS installed, which was not a problem in my case.

The issue for me was that gcc by default uses slightly different linker options on Debian/Ubuntu than on Fedora (which is what VCS is tested on). To fix this it I added `-LD_FLAGS -Wl,--no-as-needed` to the list of vcs options. That means a full command would look something like

```
vcs -full64 +v2k top.v -LDFLAGS -Wl,--no-as-needed
```

For more details on this see [this SO post](http://stackoverflow.com/questions/42113237/undefined-reference-when-linking-dynamic-library-in-ubuntu-system).

ShenShan123, 2018-07-09:
I have the same problme when make vsim. My VCS version is I-2014.03, but my host machine is 64bit platform. What should I do next?