chipsalliance/rocket-chip #165: errors occur when compiling Scala sources for Rocket

jia1546, 2016-07-15:
I run the command "make verilog" under vsim, and I get the following errors:
```
[info] Compiling 21 Scala sources to /home/meac/rocket-chip/rocket/target/scala-2.11/classes...
[error] /home/meac/rocket-chip/rocket/src/main/scala/idecode.scala:78: type mismatch;
[error]  found   : chisel3.core.UInt
[error]  required: Chisel.BitPat
[error]     (which expands to)  chisel3.util.BitPat
[error]     BNE->       List(Y,    N,N,Y,N,N,Y,Y,A2_RS2, A1_RS1, IMM_SB,DW_X,  FN_SNE,   N,M_X,      MT_X, N,N,N,N,N,N,CSR.N,N,N,N),
[error]                                          ^
[error] /home/meac/rocket-chip/rocket/src/main/scala/idecode.scala:78: type mismatch;
[error]  found   : chisel3.core.UInt
[error]  required: Chisel.BitPat
[error]     (which expands to)  chisel3.util.BitPat
[error]     BNE->       List(Y,    N,N,Y,N,N,Y,Y,A2_RS2, A1_RS1, IMM_SB,DW_X,  FN_SNE,   N,M_X,      MT_X, N,N,N,N,N,N,CSR.N,N,N,N),
```

mwachs5, 2016-07-15:
@jia1546, this command works for me.  Are you sure that all your submodules are up-to-date?

jia1546, 2016-07-17:
I use the "git pull origin master" to keep synchronous to the repository. And when I compile the scala source files to verilog under vsim sub-directory, I get a lot of errors, which is shown in the attachment. 
[compile error information.txt](https://github.com/ucb-bar/rocket-chip/files/367858/compile.error.information.txt)
@mwachs5

terpstra, 2016-07-17:
As Megan suggested, try running "git submodule update". The command you ran
only changes files in the top of the project, not the subdirectories which
are submodules.

jia1546, 2016-07-19:
After I run "git submodule update", now I error turns to an Java exception at last.
```
[info] [6.733] Done elaborating.
[success] Total time: 63 s, completed Jul 19, 2016 4:49:23 PM
mv /home/meac/rocket-chip/vsim/generated-src/Top.fir /home/meac/rocket-chip/vsim/generated-src/Top.DefaultConfig.fir
mkdir -p /home/meac/rocket-chip/vsim/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/meac/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/meac/rocket-chip/vsim/generated-src/Top.DefaultConfig.fir -o /home/meac/rocket-chip/vsim/generated-src/Top.DefaultConfig.v -X verilog
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
Exception in thread "main" java.lang.Exception: Invalid increase in scope on line 190
	at firrtl.Translator$.addBrackets(Translator.scala:118)
	at firrtl.Parser$.parse(Parser.scala:52)
	at firrtl.Driver$.compile(Driver.scala:51)
	at firrtl.Driver$.main(Driver.scala:94)
	at firrtl.Driver.main(Driver.scala)
make: *** [/home/meac/rocket-chip/vsim/generated-src/Top.DefaultConfig.v] Error 1
```

mwachs5, 2016-07-19:
Did you also make sure to re-build firrtl as mentioned in the REAMDE?

cd firrtl
sbt clean
sbt assembly

aswaterman, 2016-07-20:
Also, you have to rebuild firrtl manually at the moment ("sbt assembly"
inside firrtl/).

jia1546, 2016-07-20:
When I rebuld the firrtl, a StackOverflowError occurs: 
```
meac@ubuntu:~/rocket-chip/firrtl$ sbt assembly
[info] Loading project definition from /home/meac/rocket-chip/firrtl/project
[info] Set current project to firrtl (in build file:/home/meac/rocket-chip/firrtl/)
[info] Compiling 29 Scala sources and 4 Java sources to /home/meac/rocket-chip/firrtl/target/scala-2.11/classes...
java.lang.StackOverflowError
	at scala.tools.nsc.typechecker.Contexts$Context.makeSilent(Contexts.scala:515)
	at scala.tools.nsc.typechecker.Typers$Typer.silent(Typers.scala:675)
	at scala.tools.nsc.typechecker.Typers$Typer.normalTypedApply$1(Typers.scala:4523)
	at scala.tools.nsc.typechecker.Typers$Typer.typedApply$1(Typers.scala:4579)
	at scala.tools.nsc.typechecker.Typers$Typer.typedInAnyMode$1(Typers.scala:5342)
```

mwachs5, 2016-07-20:
I had the same problem. It depends on what version of Java you are using. I gave Java more memory by setting the SBT variable (used by the Makefiles):

> echo $SBT
java -Xmx8G -Xss512M -jar /path/to/rocket-chip/sbt-launch.jar

jia1546, 2016-07-20:
The problem is solved, and thanks for you guys! @mwachs5 @aswaterman @terpstra

terpstra, 2016-07-20:
Try: 'cd firrtl; make clean' and then compile again.

closed 2016-07-21 by hcook