chipsalliance/rocket-chip #170: weird output of C emulator

harriss95, 2016-07-18:
I did a C simulator and this is the output file for rv64ui-p-add.out

`C0:          0 [0] pc=[529b4473c8] W[r  0=0000000000000000][0] R[r 4=b9c1d835c084f7a4] R[r16=1ffebe133e9aa313] inst=[db025a65] DASM(db025a65)
C0:          1 [0] pc=[529b4473c8] W[r  0=0000000000000000][0] R[r 4=b9c1d835c084f7a4] R[r16=1ffebe133e9aa313] inst=[db025a65] DASM(db025a65)
C0:          2 [0] pc=[529b4473c8] W[r  0=0000000000000000][0] R[r 4=b9c1d835c084f7a4] R[r16=1ffebe133e9aa313] inst=[db025a65] DASM(db025a65)
C0:          3 [0] pc=[529b4473c8] W[r  0=0000000000000000][0] R[r 4=b9c1d835c084f7a4] R[r16=1ffebe133e9aa313] inst=[db025a65] DASM(db025a65)`

can anyone help me what might be the reason for this ?

terpstra, 2016-07-18:
This looks correct to me.

Does it at line 19 or so say something like:

C0:         19 [1] pc=[0000001000] W[r  0=0000000000001004][1] R[r
0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0

harriss95, 2016-07-18:
yes . that is exactly what i got at line 19

C0:         19 [1] pc=[0000001000] W[r  0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] DASM(0000006f)

can i know where u get that 

> j       pc + 0x0

terpstra, 2016-07-18:
If you pipe the output through spike-dasm it will show you the instructions
disassembled.

closed 2016-07-21 by hcook