chipsalliance/rocket-chip #174: Can't generate ZynqAdapter.DefaultFPGAConfig.v

zpod86300, 2016-07-21:
I run the command "make rocket" under fpga-zynq/zedboard.

When it run to the 40 line in the fpga-zynq/common/Makefrag :
   make verilog MODEL=ZynqAdapter CONFIG=$(CONFIG); \

It will occur the error.

```
cd /home/test/Program/rocket-chip && java -Xmx4G -Xss512M -XX:MaxPermSize=256M -jar /home/test/Program/rocket-chip/sbt-launch.jar "run rocketchip ZynqAdapter DefaultFPGAConfig --W0W --minimumCompatibility 3.0.0 --backend v --configName DefaultFPGAConfig --compileInitializationUnoptimized --targetDir /home/test/Program/rocket-chip/fsim/generated-src --configDump --noInlineMem"
[info] Loading project definition from /home/test/Program/rocket-chip/project
Using addons: 
[info] Set current project to rocketchip (in build file:/home/test/Program/rocket-chip/)
[info] Running rocketchip.TestGenerator rocketchip ZynqAdapter DefaultFPGAConfig --W0W --minimumCompatibility 3.0.0 --backend v --configName DefaultFPGAConfig --compileInitializationUnoptimized --targetDir /home/test/Program/rocket-chip/fsim/generated-src --configDump --noInlineMem
[info] [0.004] Elaborating design...
[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
	at rocketchip.TestGenerator$$anonfun$23.apply(Testing.scala:182)
	at rocketchip.TestGenerator$$anonfun$23.apply(Testing.scala:179)
	at chisel3.core.Module$.do_apply(Module.scala:30)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:111)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:111)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:121)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:119)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:119)
	at chisel3.Driver$.elaborate(Driver.scala:111)
Caused by: java.lang.IllegalArgumentException: requirement failed: TODO reimplement using debug port, not HTIF
	at scala.Predef$.require(Predef.scala:219)
	at rocketchip.ZynqAdapter.<init>(Fpga.scala:24)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
	at rocketchip.TestGenerator$$anonfun$23.apply(Testing.scala:182)
	at rocketchip.TestGenerator$$anonfun$23.apply(Testing.scala:179)
	at chisel3.core.Module$.do_apply(Module.scala:30)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:111)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:111)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:121)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:119)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2077)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:750)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:748)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:run) Nonzero exit code: 1
[error] Total time: 3 s, completed Jul 21, 2016 2:30:25 PM
make[1]: Leaving directory `/home/test/Program/rocket-chip/fsim'
cp: cannot stat ‘generated-src/{Top,ZynqAdapter}.DefaultFPGAConfig.v’: No such file or directory
make: *** [rocket] Error 1
```

sbeamer, 2016-07-21:
Are you using bash as your shell?

If not, the next fpga-zynq commit after the one you are using should fix this. You can get it by simply moving into the fpga-zynq directory and checking out master.

Please note that fpga-zynq is currently not compatible with the head of master of rocket-chip, and it may take a substantial amount of time until full support is restored. We believe 255ef05e2 is the most recent rocket-chip repo commit that is compatible.

zpod86300, 2016-07-21:
I had added the following code in the .bashrc and I never change the path of all file in rocket-chip.

```
export TOP=/home/test/Program
export RISCV=$TOP/riscv
export ROCKETCHIP=$TOP/rocket-chip
export REPO=$TOP/fpga-zynq
export PATH=$PATH:$RISCV/bin
```

Is the shell in .bashrc still not enough to make rocket?
Really appreciate your help!

zpod86300, 2016-08-13:
I update my rocket-chip generator and it still not work correctly.
The error code is following.

```
make[1]: Entering directory `/home/test/Program/rocket-chip/vsim'
mkdir -p /home/test/Program/rocket-chip/vsim/generated-src/
cd /home/test/Program/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/test/Program/rocket-chip/sbt-launch.jar "run rocketchip ZynqAdapter DefaultFPGAConfig --W0W --minimumCompatibility 3.0.0 --backend v --configName DefaultFPGAConfig --compileInitializationUnoptimized --targetDir /home/test/Program/rocket-chip/vsim/generated-src --configDump --noInlineMem"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /home/test/Program/rocket-chip/project
Using addons: 
[info] Set current project to rocketchip (in build file:/home/test/Program/rocket-chip/)
[info] Running rocketchip.RocketChipGenerator rocketchip ZynqAdapter DefaultFPGAConfig --W0W --minimumCompatibility 3.0.0 --backend v --configName DefaultFPGAConfig --compileInitializationUnoptimized --targetDir /home/test/Program/rocket-chip/vsim/generated-src --configDump --noInlineMem
[info] [0.003] Elaborating design...
[error] (run-main-0) java.lang.ClassNotFoundException: rocketchip.ZynqAdapter
java.lang.ClassNotFoundException: rocketchip.ZynqAdapter
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at rocketchip.GeneratorUtils$$anonfun$1.apply(Generator.scala:35)
	at rocketchip.GeneratorUtils$$anonfun$1.apply(Generator.scala:34)
	at chisel3.core.Module$.do_apply(Module.scala:30)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:111)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:111)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:130)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:128)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:128)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2077)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:750)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$38$$anonfun$apply$39.apply(Defaults.scala:748)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:run) Nonzero exit code: 1
[error] Total time: 3 s, completed Aug 13, 2016 8:27:31 PM
make[1]: Leaving directory `/home/test/Program/rocket-chip/vsim'
cp: cannot stat ‘generated-src/ZynqAdapter.DefaultFPGAConfig.v’: No such file or directory
make: *** [rocket] Error 1
```

It say don't find class "rocketchip.ZynqAdapter".
How do I solve this problem?

The line 39 of file "Makefrag" in fpga-zynq/common, I think it should be changed to "cd $(base_dir)/rocket-chip/vsim; \ ".

davidbiancolin, 2016-08-13:
To reiterate what Scott said, the most recently releases of rocket simply will not work with an older version of fpga-zynq. For instance, the source for that adapter no longer lives in rocket-chip. Updating rocket-chip will only break things further. 

In any case, checkout out a clean version of fpga-zynq may resolve your problems, as i have recently updated much of the source, but not the images, to support generation along side the most recent versions of rocket-chip. If you continue to have issues, open an issue there: (https://github.com/ucb-bar/fpga-zynq)

closed 2016-08-13 by davidbiancolin