chipsalliance/rocket-chip #179: Rocket non-blocking cache loses result in back-to-back write

zhemao, 2016-07-28:
This problem came up when using the trace generator with the non-blocking L1 DCache. If two writes to different blocks with the same index go into the cache back-to-back and the first write was a hit on a previously clean line, the result of the first write will be lost.

The issue is that the first write updates the metadata state to dirty. However, the second write reads the metadata before this update occurs, believes that it can evict the block without writeback (since it's clean), and overwrites the change made by the first write.

AXE reports failing case with random seed 3 using TraceGenL2Config with L1D_MSHRS set to 0 and NTiles set to 1. Shrunken AXE trace as follows.

    0: { M[3] == 0; M[3] := 32} @ 489:
    0: M[3] := 54 @ 904:
    0: M[3] == 32 @ 1092:1172

Waveform is on i0 under /home/zhemao/tracegen.vcd.

aswaterman, 2016-08-01:
resolved by 8a7fc75b5377f891390ed4f5bd2fcd99510c4bcb

closed 2016-08-01 by aswaterman

sammy17, 2022-07-25:
Hi, I am trying to reproduce this bug by inserting it into the latest Rocket core source for an evaluation of a verification tool.
I tried reverting the fix by removing s2_update_meta from the line: https://github.com/chipsalliance/rocket-chip/blob/b503f8ac28a497b2463ffbac84bfe66533ace0bb/src/main/scala/rocket/DCache.scala#L419  

I also created a program that writes two consecutive writes to the same cache index, I made sure the first write is a hit on a clean line. However, I am not seeing the same issue reported by the bug. 
Please let me know if my understanding of the scenario that triggers the bug is incorrect. 

**Could anyone please help with reproducing this issue with the latest Rocket core?**