chipsalliance/rocket-chip #180: riscv-tools needs a bump

ekrimer, 2016-07-28:
There is an issue with current version of riscv-tools (7219be435a89277603e566e806ae8540c7f9a917).

$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c
$ riscv64-unknown-elf-gcc -o hello hello.c
$ spike pk hello

The sequence above would currently result in a segmentation fault. 

Described here : https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/YK7xA9sfVZc

Turns out that updating to latest riscv-tools (816a2527a0c2181a5d2150b3412399b523afc106) solves this issue.

Can riscv-tools be bumped please? Thanks.

zhemao, 2016-07-28:
7219be435a89277603e566e806ae8540c7f9a917 is the latest version of riscv-tools. The commit you see on master is for the previous priv-1.7 ISA spec. It will not work with the current rocket-chip master.

That sequence of commands works for me with no segfault. I'm using the same riscv-tools commit. What operating system are you running on?

ekrimer, 2016-07-28:
I am on CentOS release 5.6. Building with gcc 4.9.2. 

Getting segfault with 7219be435a89277603e566e806ae8540c7f9a917 and it works nicely with 816a2527a0c2181a5d2150b3412399b523afc106. 

I've posted all the details of my poking so far in riscv sw-dev post linked above.

Any suggestions?

zhemao, 2016-07-30:
The loop you are seeing it stuck on is how the processor tells the frontend server that it has finished. So it's likely the segfault is from the frontend server. Not sure how you would go about debugging this. Maybe try building frontend server with debug symbols on and then run spike in gdb?

ekrimer, 2016-09-02:
I've got to debug the issue and submitted a pull request : https://github.com/riscv/riscv-fesvr/pull/15

closed 2016-09-07 by hcook