chipsalliance/rocket-chip #183: Rocket frontend mask logic change of behavior

ccelio, 2016-07-30:
Introduced by commit https://github.com/ucb-bar/rocket/commit/d91be2b545ea0eb704bf3c2b02b5d200f27426fa#diff-dbf1307484c9f6974955adde94a99967, the behavior of the masking logic has changed (https://github.com/ucb-bar/rocket-chip/blob/master/rocket/src/main/scala/frontend.scala#L128):

````scala
io.cpu.resp.bits.mask := UInt((1 << fetchWidth)-1) << s2_pc.extract(log2Up(fetchWidth)+log2Up(coreInstBytes)-1, log2Up(coreInstBytes))
````

For `fetchWidth=1`, `coreInstBytes=4`, that reduces to

> mask := UInt(1) << s2_pc(1+2-1, 2)
> mask := 1 << pc(2,2)

This means `mask` can be 0.

The old behavior, for `fetchWidth = 1`, was for `mask` to be set to UInt(1) (as the one instruction will always be valid if imem.valid is true).

Is the changed behavior intentional?

aswaterman, 2016-07-30:
> Introduced by commit ucb-bar/rocket@d91be2b
> #diff-dbf1307484c9f6974955adde94a99967

No; the problem is log2Up (it would have noisily failed using log2Ceil).
Will fix.

ccelio, 2016-07-30:
While I have you, what's the meaning of "extract"? I assume this has different semantics from "()"; does it handle (-1,0)?

aswaterman, 2016-07-31:
It's a widget I added to rocket utils, which returns UInt(0) when
attempting to extract (n-1, n).

aswaterman, 2016-08-01:
This will be fixed by a6e009d8debf3a4a058275f8de9fc270e68ddfa2 once it gets merged

closed 2016-08-01 by aswaterman