chipsalliance/rocket-chip #184: Rocket BTB bridx logic change of behavior

ccelio, 2016-07-30:
A recent commit to Rocket changed the behavior of the BTB's bridx logic.

https://github.com/ucb-bar/rocket-chip/blob/master/rocket/src/main/scala/btb.scala#L158

For a `FetchWidth=1`, the btb's resp.bits.bridx can now return 1, which is out-of-bounds. 

The problem is:
* the bridx vector is not initialized (understandably, of course).
* the bridx vector is not written to if fetchWidth = 1.

This means that bridx can return 1.  The fix is to hardwire `resp.bits.bridx` to 0 if FetchWidth=1.

aswaterman, 2016-07-31:
I believe 058396aefe4103046406be63e1a4c0178c4d13b3 fixed this regression.  Can you confirm?

ccelio, 2016-08-01:
It looks like this line should probably fix it: https://github.com/ucb-bar/rocket-chip/commit/058396aefe4103046406be63e1a4c0178c4d13b3#diff-4ad90e4d8a82ac845775321ea2c06141R236, so I'll mark this closed.

Unfortunately, I think it will take a bit of work before I can begin to wrap my head around that commit and how it affects superscalar BOOM.

closed 2016-08-01 by ccelio