chipsalliance/rocket-chip #188: RoCC L1/L2 Reads/Writes Triggering Rocket/Uncore Assertions

seldridge, 2016-08-03:
After moving to Chisel3 and integrating everything that's happened over the past ~2 months, I've started to see failures for my RoCC accelerator when doing L2 reads. I switched back to using a basic test program that exercises the provided [`AccumulatorExample`](https://github.com/ucb-bar/rocket-chip/blob/master/rocket/src/main/scala/rocc.scala#L73) accelerator. This is tripping an assertion when trying to do L1 reads.

Additionally, while I don't have this integrated with [`test-accumulator`](https://github.com/ucb-bar/rocket-chip/blob/master/uncore/src/main/scala/tilelink/Interconnect.scala#L219), I have a [debug unit as part of a larger accelerator](https://github.com/bu-icsg/xfiles-dana/blob/master/src/main/scala/XFilesDebugUnit.scala#L12) that I can command to exercise the RoCC port via a [test program](https://github.com/bu-icsg/xfiles-dana/blob/master/src/test/rv/debug-test.c). I'm seeing the same failure on that for L1 reads/writes, a different assertion firing for L2 reads, and L2 writes don't appear to be getting data where they need to go. 

I fully realize that this may all be on my end, either in the software or that rocket/uncore have moved out from under me. I can do some more digging tomorrow as well as try and get at what these assertions actually mean.

### L1 Reads
I'm running [`test-accumulator`](https://github.com/seldridge/rocket-rocc-examples/blob/master/src/test/c/test-accumulator.c) which initiates a `doLoad` request on the address of a data value (0xbeef stored in 0xfeefae8). Output below:
```
...
[INFO] Load beef (0xfeefae8) via L1 data cache
[INFO]   Reading back in software [0xfeefae8] = beef
[2240987] %Error: Top.RoccExampleConfig.v:57842: Assertion failed in TOP.Top.tileList_0.SimpleHellaCacheIF_3
%Error: /home/se/research_local/rocket-chip-clean/emulator/generated-src/Top.RoccExampleConfig.v:57842: Verilog $stop
Aborting...
```
That specific failure is [`nbdcache.scala`](https://github.com/ucb-bar/rocket-chip/blob/master/rocket/src/main/scala/nbdcache.scala#L1243)

### L1 Writes
I'm seeing the same assertion firing as for L1 Reads.

### L2 Reads
This is tripping an assertion in [`Interconnect.scala`](https://github.com/ucb-bar/rocket-chip/blob/master/uncore/src/main/scala/tilelink/Interconnect.scala#L219).

### L2 Writes
This succeeds, but doesn't seem to be getting the correct data written.

zhemao, 2016-08-03:
Are you sure you're reading and writing the correct addresses? The position of memory in the address map has changed. It no longer starts at 0. The addresses you read and write to should be after 0x80000000.

zhemao, 2016-08-03:
OK, I see what the problem is here. The issue is that pk is no longer using direct mapping. The accumulator uses physical addresses, but you are passing it a virtual address. So it does a load from physical address 0xfeefae8, which is not a valid part of the address map (memory only ranges from 0x80000000 - 0x90000000).

I modified your code to use the translator accelerator to get the physical address and pass that to the accumulator to perform the load. It seems to work fine now.

closed 2016-08-03 by zhemao

seldridge, 2016-08-03:
Thanks, @zhemao. Yes, pk decided to stop using physical addresses (looks like over a month ago) and that was breaking this. Thanks for the PR!