chipsalliance/rocket-chip #203: Newest version failed to compile emulator on mac

hauhsu, 2016-08-17:
Hi, 
I cloned the newest rocket-chip repo and 
follow the instructions to compile riscv-tools and emulator. 
It worked fine on my iMac about a month ago,
but the current version failed while compiling emulator.

The following are the error messages: 

```
[info] Compiling 24 Scala sources to /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/target/scala-2.11/classes...
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/nbdcache.scala:10: object util is not a member of package uncore
[error] import uncore.util._
[error]               ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/icache.scala:6: object util is not a member of package uncore
[error] import uncore.util._
[error]               ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/dcache.scala:10: object util is not a member of package uncore
[error] import uncore.util._
[error]               ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/dcache.scala:164: not found: type StoreGen
[error]   val misaligned = new StoreGen(s1_req.typ, s1_req.addr, UInt(0), wordBytes).misaligned
[error]                        ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/dcache.scala:190: not found: type StoreGen
[error]   val pstore1_storegen = new StoreGen(pstore1_typ, pstore1_addr, pstore1_data, wordBytes)
[error]                              ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/dcache.scala:400: not found: type LoadGen
[error]   val loadgen = new LoadGen(s2_req.typ, mtSigned(s2_req.typ), s2_req.addr, s2_data_word, s2_sc, wordBytes)
[error]                     ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/dcache.scala:407: not found: type AMOALU
[error]     val amoalu = Module(new AMOALU)
[error]                             ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/icache.scala:74: not found: value FlowThroughSerializer
[error]   val narrow_grant = FlowThroughSerializer(io.mem.grant, refillCyclesPerBeat)
[error]                      ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/nbdcache.scala:189: not found: type StoreGen
[error]   val storegen = new StoreGen(req.typ, req.addr, req.data, wordBytes)
[error]                      ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/nbdcache.scala:190: not found: type LoadGen
[error]   val loadgen = new LoadGen(req.typ, mtSigned(req.typ), req.addr, grant_word, req_cmd_sc, wordBytes)
[error]                     ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/nbdcache.scala:869: not found: type StoreGen
[error]   val misaligned = new StoreGen(s1_req.typ, s1_req.addr, UInt(0), wordBytes).misaligned
[error]                        ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/nbdcache.scala:962: not found: type AMOALU
[error]   val amoalu = Module(new AMOALU)
[error]                           ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/nbdcache.scala:1017: not found: value FlowThroughSerializer
[error]   val narrow_grant = FlowThroughSerializer(io.mem.grant, refillCyclesPerBeat)
[error]                      ^
[error] /Users/hhsu/Projects/riscv/rocket-chip-test/rocket/src/main/scala/nbdcache.scala:1065: not found: type LoadGen
[error]   val loadgen = new LoadGen(s2_req.typ, mtSigned(s2_req.typ), s2_req.addr, s2_data_word, s2_sc, wordBytes)
[error]                     ^
[error] 14 errors found
[error] (rocket/compile:compileIncremental) Compilation failed
[error] Total time: 68 s, completed Aug 17, 2016 9:34:46 PM
```

The OS version is EL Capitan 10.11.6
I also tried to compile on Ubuntu 14.04, no errors.
Can anyone help? 
Thanks.

zhemao, 2016-08-22:
I wonder if this is a java issue. Which version of java are you running?

hauhsu, 2016-08-23:
```
$ java -version
java version "1.8.0_71"
Java(TM) SE Runtime Environment (build 1.8.0_71-b15)
Java HotSpot(TM) 64-Bit Server VM (build 25.71-b15, mixed mode)
```

But the issue disappeared after I pull a newer version....
Should I close it?

closed 2016-08-23 by hauhsu