chipsalliance/rocket-chip #221: Should we tie invalidate_lr to false for RoCC accelerator?

zhemao, 2016-08-23:
I got bitten by this today. If you are using a RoCC accelerator, you need to remember to tie io.mem.invalidate_lr to false or else LR/SC instructions will continuously fail.

Since this keeps happening, perhaps we should tie this to false in the RoCC module. I can't think of a case in which you would actually want to send LR/SC from a RoCC accelerator. We already tie io.mem.req.bits.phys to false, so it's not like there isn't precedent for this.

aswaterman, 2016-08-23:
Probably so.

LR/SC would rarely make sense, since there's only a single load reservation register in the D$.  So the core and the RoCC could livelock each other.

colinschmidt, 2016-08-24:
I have no problem adding this to the RoCC module (Hwacha ties it to false). I also think that anyone using the RoCC interface just kind of needs to keep up with Rocket changes. Its not a standard interface and it changes often, so its always going to have issues like this.

zhemao, 2016-08-26:
I'll put together a PR for this in a bit.

closed 2016-09-01 by zhemao