chipsalliance/rocket-chip #223: emulator output for NTILES=2

harriss95, 2016-08-25:
This is the output from rv64ui-p-add.out , it seems that C1 is not working . Is this how it is supposed to be ?

> 
C1:      27166 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27167 [1] pc=[00800000bc] W[r 1=0000000000000000][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000093] li      ra, 0
C1:      27167 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27168 [1] pc=[00800000c0] W[r 2=0000000000000000][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000113] li      sp, 0
C1:      27168 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27169 [1] pc=[00800000c4] W[r 3=0000000000000000][1] R[r 1=0000000000000000] R[r 2=0000000000000000] inst=[002081b3] add     gp, ra, sp
C1:      27169 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27170 [1] pc=[00800000c8] W[r29=0000000000000000][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000e93] li      t4, 0
C1:      27170 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27171 [1] pc=[00800000cc] W[r28=0000000000000002][1] R[r 0=0000000000000000] R[r 2=0000000000000003] inst=[00200e13] li      t3, 2
C1:      27171 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27172 [1] pc=[00800000d0] W[r 0=0000000000000000][0] R[r 3=0000000000000000] R[r29=0000000000000000] inst=[4fd19063] bne     gp, t4, pc + 1248
C1:      27172 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27173 [1] pc=[00800000d4] W[r 1=0000000000000001][1] R[r 0=0000000000000000] R[r 1=0000000000000003] inst=[00100093] li      ra, 1
C1:      27173 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27174 [1] pc=[00800000d8] W[r 2=0000000000000001][1] R[r 0=0000000000000000] R[r 1=0000000000000001] inst=[00100113] li      sp, 1
C1:      27174 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:      27175 [1] pc=[00800000dc] W[r 3=0000000000000002][1] R[r 1=0000000000000001] R[r 2=0000000000000001] inst=[002081b3] add     gp, ra, sp

This is the output from rv64ui-v-add.out , it seems that C1 is not working . Is this how it is supposed to be ?

> C0:     140723 [1] pc=[0000001a00] W[r 1=0000000000000000][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000093] li      ra, 0
C1:     140723 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:     140724 [1] pc=[0000001a04] W[r 2=0000000000000000][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000113] li      sp, 0
C1:     140724 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0
C0:     140725 [1] pc=[0000001a08] W[r 3=0000000000000000][1] R[r 1=0000000000000000] R[r 2=0000000000000000] inst=[002081b3] add     gp, ra, sp
C1:     140725 [1] pc=[0000001000] W[r 0=0000000000001004][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[0000006f] j       pc + 0x0

colinschmidt, 2016-08-25:
The assembly tests are written such that only core/hart 0 executes the test. So this output for C1 is expected. See https://github.com/riscv/riscv-test-env/blob/master/p/riscv_test.h#L76

harriss95, 2016-08-26:
Is there a way to enable multicore for assembly test ?

zhemao, 2016-08-26:
It isn't really feasible to have the assembly tests run on multiple cores. It might be possible for the virtual memory tests by giving them separate address maps, but for the physical memory tests it just isn't possible. Both cores will be executing in the same memory space, overwriting each others' data.

closed 2016-08-29 by harriss95