chipsalliance/rocket-chip #232: Heterogeneous core configs can not generate the correct config string

colinschmidt, 2016-08-30:
The way the config string is created currently only looks at one of the core's parameters for things like what isa is supported.
rocket-chip should be able to handle the creation of heterogeneous ISA cores including the correct config string.

aswaterman, 2016-08-30:
This is already on the docket.

colinschmidt, 2016-08-30:
Cool. It would also be useful if the RoCCParameters included an extension name. Then just add something like the following to the isa string as well.

```
${site(BuildRoCC).map(_.xName).mkString}
```

aswaterman, 2016-08-30:
Agreed.  We plan to standardize how components register themselves in the config string & address map (it's quite ad hoc now); RoCC should use that mechanism once it exists.

closed 2017-03-24 by hcook