chipsalliance/rocket-chip #233: Typo in SimAXIMem?

pbing, 2016-08-30:
TestHarness.scala:145
```
 io.axi.r.bits.data := mem((ar.addr >> log2Ceil(w.data.getWidth/8))(log2Ceil(depth)-1, 0))
```
Should it be
```
 io.axi.r.bits.data := mem((ar.addr >> log2Ceil(r.data.getWidth/8))(log2Ceil(depth)-1, 0))
```
instead?

zhemao, 2016-08-30:
Probably, but in AXI, the write and read data widths are the same, so this is still correct.

aswaterman, 2016-08-30:
Yeah, I'll fix it, though... at the very least it is confusing.

aswaterman, 2016-08-30:
Looks like I did it on purpose to avoid having to write 

    io.axi.r.bits.data.getWidth/8

zhemao, 2016-08-30:
Couldn't we just use nastiXDataBits?

aswaterman, 2016-08-30:
yeah that would make sense

pbing, 2016-09-04:
I think wValid should be renamed to wReady (axi.w.valid is driven by master).

```
  val wReady = Reg(init = Bool(false))
  val bValid = Reg(init = Bool(false))
  val aw = RegEnable(io.axi.aw.bits, io.axi.aw.fire())
  io.axi.aw.ready := !wReady && !bValid
  io.axi.w.ready := wReady
  when (io.axi.b.fire()) { bValid := false }
  when (io.axi.aw.fire()) { wReady := true }
  when (io.axi.w.fire()) {
    assert(aw.burst === NastiConstants.BURST_INCR)
    aw.addr := aw.addr + (UInt(1) << aw.size)
    aw.len := aw.len - 1
    when (aw.len === UInt(0)) {
      wReady := false
      bValid := true
    }
```

aswaterman, 2016-09-04:
in this context, wValid means "a transaction on the w channel is being processed", and has nothing to do with direction

closed 2016-09-04 by aswaterman