chipsalliance/rocket-chip #238: Enhance PRCI to handle Clock Configuration

mwachs5, 2016-09-01:
PRCI is supposed to handle the Clock Configuration logic for Rocket-Chip, but currently it doesn't do much relating to clocks. 

I'd like to enhance PRCI to include registers for Clock Configuration (vs. creating a new peripheral that does Clock Configuration/ Status tasks). The actual interpretation of these values would happen outside of PRCI and the Coreplex. 

This is completely design specific, so PRCI would need to take a parameter for these ports, similar to p(ExtraCoreplexPorts). I'd create p(ClockConfig) which is passed into PRCI to create registers which can be read & written by TileLink, and create ports which punch out of Coreplex (and Top, by default).

aswaterman, 2016-09-02:
I have a feeling that PRCIs will differ enough that creating new chip-specific peripherals will prove less unwieldy than passing stuff into the one true PRCI, but we won't know til we try, so we can give your way a shot.

mwachs5, 2016-09-02:
I think you're right. I was trying to squeeze this into the current thing.

Because PRCI lives within Coreplex, it's hard to change it without writing
a new BuildCoreplex function. Maybe a better plan would be to make PRCI a
parameter so that the whole PRCI peripheral could be overridden.

mwachs5, 2016-09-06:
For now, I've decided to just create a different peripheral that handles all this.

closed 2016-09-15 by hcook