chipsalliance/rocket-chip #249: Can't set NExtTopInterrupts to 0

ben-k, 2016-09-07:
This triggers an assertion, but it ought to be a perfectly reasonable thing to do - or else some other option for disabling external interrupts entirely.

mwachs5, 2016-09-07:
Which assertion failure?

I assume you mean that both NExtTopInterrupts + NExtPeripheryInterrupts = 0. This is checked by PRCI, I thought there was already an issue raised about this... (though hard to find: 
https://github.com/ucb-bar/fpga-zynq/pull/15#issuecomment-239583084)

Or do you really mean that NExtTopInterrupts = 0, but NExtPeripheryInterrupts > 0 causes an assertion failure? If so I will take a look.

ben-k, 2016-09-07:
Thanks - I should have clarified.  When I build DefaultConfig but set `NExtTopInterrupts = 0`, I get the following assertion:

```
Caused by: java.lang.IllegalArgumentException: requirement failed
       	at scala.Predef$.require(Predef.scala:207)
       	at uncore.devices.PLICConfig.<init>(Plic.scala:54)
```

I had seen the discussion in fpga-zynq, but the resolution there seemed to have been a workaround, not a fix.  In our case, we don't want to just tie the pins off in the testharness - we actually want to eliminate them to save on top-level IOs, which are a scarce resource in ASIC-land.

As you suggest, it looks like the requirement demands a single interrupt from anywhere - so I guess I can work around this by adding a periphery interrupt and then tying it off.  But what's the correct solution here?

mwachs5, 2016-09-07:
I think the correct solution is probably to change PRCI to be OK with not
having any interrupts.

mwachs5, 2016-09-07:
I mean PLIC. Gets me every time.

aswaterman, 2016-09-07:
I'm fixing it now.

commit 02a24392 (2016-09-07): Support a degenerate PLIC with no interrupts

Resolves #249
closed 2016-09-07 by aswaterman