chipsalliance/rocket-chip #251: Executing an "rdtime" instruction in bbl gives an illegal instruction exception

sashimi-yzh, 2016-09-07:
The newest version of riscv-linux use "rdtime" instruction to read the time counter. But I can not boot riscv-linux over the rocket-chip. I try to insert an "rdtime" instruction in bbl before switching to supervisor mode. It gives me an illegal instruction exception. The binary of this instruction is
```
800002d0:   c01025f3            rdtime  a1
```

Is the "rdtime" instruction supported by the current version of rocket-chip?

aswaterman, 2016-09-07:
Read the mtime global instead.

rdtime is emulated for less privileged software and illegal within bbl
itself.

sashimi-yzh, 2016-09-08:
Can "rdtime" be used in supervisor mode? The priv-1.9 branch of riscv-linux use "rdtime", but I get an illegal instruction exception when riscv-linux executes rdtime.

aswaterman, 2016-09-08:
When S/U-mode software executes it, M-mode software (e.g., bbl) should trap
it and emulate it.

sashimi-yzh, 2016-09-10:
The following pull request fixed this issue.

https://github.com/riscv/riscv-pk/pull/37

closed 2016-09-10 by sashimi-yzh