chipsalliance/rocket-chip #256: Add test to regression for MMIO + Scratchpad

mwachs5, 2016-09-08:
There is an issue with routing to the external MMIO port if you also have a Scratchpad. @yunsup  is working on a fix, but we should also add a test to the regression for this case.

zhemao, 2016-09-13:
Really, we need more tests of MMIO in general. I guess I can add a cache regression test which tries to write out to MMIO passing through the L1.

zhemao, 2016-09-13:
So what's the bug? If you have a scratchpad and try to make an MMIO request, what happens?

mwachs5, 2016-09-13:
This was fixed already by commit 8536a2a But it wasn't caught in the regression.

mwachs5, 2016-09-13:
The bug there was that if you had a scratchpad at 0x8000_0000, because it was in 'int:' memory region, the 'int' memory region spanned the MMIO devices at say 0x5000_0000, so at runtime you'd get 'no possible route' assertion failure.

terpstra, 2016-11-23:
This has been fixed by replacing all the TL1 crossbars with TL2. The scratchpad is routed just like any other device.

closed 2016-11-23 by terpstra