chipsalliance/rocket-chip #267: emulator doesn't set STOP_COND

colinschmidt, 2016-09-09:
Currently when you build your design in the emulator it does not set the STOP_COND variable so any asserts that occur during reset will cause the simulation to fail.

In vsim STOP_COND and PRINTF_COND are both set to values from the TestDriver, which does not exist in emulator.

colinschmidt, 2016-09-09:
I'm going to look into this a little bit but if anyone else has a better idea of what the solution should be let me know.

colinschmidt, 2016-09-09:
Ok it actually wasn't difficult. submitting a PR as soon as the tests pass.

aswaterman, 2016-09-09:
I take it you're handling it similarly to how the emulator handles PRINTF_COND?

zhemao, 2016-09-15:
This got fixed right? Can I close the issue?

colinschmidt, 2016-09-15:
Yup.

closed 2016-09-15 by colinschmidt