chipsalliance/rocket-chip #281: Fuzzer needs updating for new Queue and IrrevocableIO semantics

ucbjrl, 2016-09-13:
rocket-chip tests are failing with:

    [error] /vm/home/jenkins/workspace/rocket-chip_with_chisel3/src/main/scala/uncore/tilelink2/Fuzzer.scala:131: value irrevocable is not a member of object chisel3.util.Queue
    [error]     val alloc = Queue.irrevocable(idMap.io.alloc, 1, pipe = true)
    [error]                       ^
    [error] /vm/home/jenkins/workspace/rocket-chip_with_chisel3/src/main/scala/uncore/tilelink2/Fuzzer.scala:131: not found: value pipe
    [error]     val alloc = Queue.irrevocable(idMap.io.alloc, 1, pipe = true)
    [error]                                                      ^

hcook, 2016-09-13:
I'm confused by this. Our CI tests compile and pass. How is jenkins handling submodule pointer bumps to chisel3 within rocket-chip? PR #283 bumps it again just to be sure.

zhemao, 2016-09-13:
@ucbjrl did you remember to `git submodule update`? The chisel3 submodule pointer has changed.

ucbjrl, 2016-09-14:
Jenkins artificially builds rocket-chip with the most recent version of Chisel, so this isn't a bug yet.

We're trying to ensure that changes to Chisel (or firrtl), don't unknowingly break rocket-chip. 

I shouldn't have filed this as an issue. I'll close it. (I have too many balls in the air today.)

ucbjrl, 2016-09-14:
We don't encounter this with the version of Chisel that's currently used to build rocket-chip.

closed 2016-09-14 by ucbjrl