chipsalliance/rocket-chip #317: eliminate hardcoded debugrom contents from Debug.scala

ekrimer, 2016-09-21:
There are hardcoded debugrom contents in src/main/scala/uncore/devices/Debug.scala coming from riscv-tools/.... 

I am thinking how to have an easier update cycle for debugrom. I could set it up to pull content from appropriate img files from $RISCV... I think, but then I lose make dependency.

Any thoughts on the best way to organize this? Maybe to have a local copy of debug-rom .S and process it similarly to bootrom? Then there is an issue of possible diverging from spike debug rom, but on the other hand, anyway spike is a little different, and it is up to the user to keep these in sync if that is important.

What are your thoughts?

aswaterman, 2016-09-21:
Incidentally, we're already working on doing something along those lines.

mwachs5, 2016-09-21:
Have they already fallen out of sync, or is this a hypothetical?

ekrimer, 2016-09-21:
@mwachs5 I don't think I noticed them being out of sync, although currently with hardcoded content nothing prevents from getting out of sync. I mainly want to have an easy way to update debugrom. 

@aswaterman :+1:  Glad I asked. I'll then hold off and wait for your changes. Let me know if I can help.

aswaterman, 2016-09-21:
@ekrimer just to expound, we're working on doing this for the boot ROM, which will serve as a template for how to do it for the debug ROM.  We'll generalize it & port it to the debug ROM thereafter.

ekrimer, 2016-09-21:
Doesn't bootrom already pull content from file configured by `case BootROMFile => "./bootrom/bootrom.img"` ?

aswaterman, 2016-09-21:
Only partially.  There's also some boot ROM contents that are spat out of the RTL generator, which are merged together awkwardly with the contents of the boot ROM.

We wanted to clean up that approach before fixing the debug ROM, because in general, the debug ROM contents should be linked based upon output of the RTL generator.

aswaterman, 2016-09-21:
The net effect is that an ELF will be linked after (and as a result of) the RTL generator executing, in both cases.

ekrimer, 2016-09-21:
Thanks for clarification.

Actually to followup on divergence point, spike and rocket are actually using different binaries of debug rom. Spike uses only the "debug_rom_raw" version which is compiled with DRV32 and DRV64. Rocket does have that image as "defaultRomContents" but never uses it. Rocket uses either 32- or 64-bit versions that are never used by spike.

Can we start with getting rid of defaultRomContents array in rocket which is never used but confuses (it got me :) ) - https://github.com/ucb-bar/rocket-chip/pull/318

mwachs5, 2016-09-21:
Well, I'm not sure we want to get rid of it. Someone might want to use those in the future (there are many many things described in the RocketChip configurations that "no one" uses, but I know are being used), as rocket-chip has turned into a sort of library.

Maybe a name change from "Default" would be better.

ekrimer, 2016-09-21:
I can see your argument if it was potentially reachable by config, and there is just no config to reach it. For example if in config we'd specify explicitly the name of debug rom to use. Right now it is a dead code with a binary blob, that can't be reached or regressed.  

Not urgent as sounds like @aswaterman is going to clean up the whole mechanism anyway, but in general I think it is a good idea to eliminate dead code that can not be accessed and or regressed. If you think there are configs that are using it, it should at least be accessible by config, and better regressed.

mwachs5, 2016-10-01:
@ekrimer missed your response until now. 
Rather than getting rid of the contents we should identify the case in which we'd need the "Default" and write the code for that. In general you'd need the "Default" version if you have a system with both RV32 and RV64 cores (RV128 is a TODO item, I believe). If rocket-chip supports creating such a system, we should determine that and instantiate that version of the ROM.

closed 2017-03-24 by hcook