chipsalliance/rocket-chip #331: Update documentation in README.md

zhemao, 2016-09-22:
After the cake pattern refactor, the documentation on "The Top Level Module" is completely outdated. After things settle down, we should take the time to fix it so that it accurately reflects the state of the codebase. Probably instead of talking about what we've done specifically, it would be more helpful to instruct users on how to piece together their own top-level using traits.

mwachs5, 2016-10-31:
I started working on this, but it's a quite a diff. My basic plan of attack:

* Rewrite "The Top Level Module" explanation assuming that you just did "make verilog CONFIG=DefaultConfig", and just describing what the verilog I/Os mean.

* There are now two real ways to "how can I parameterize my rocket-chip": (1) The old way using CONFIG, and (2) the new way, mixing traits to make your own BaseTop (or ExampleTop) subclass.

*  Basic explanation of the 3-piece cake pattern repeated all over the place now (lazy module definition, Module I/O definitition, module elaboration), explain some part of ExampleTop that shows how that 3-piece pattern works.

zhemao, 2016-10-31:
If it helps, I wrote an example of using the 3-piece cake pattern to create a custom top. It's in the README of the [project template](https://github.com/ucb-bar/project-template) repository.

mwachs5, 2016-11-01:
That is super helpful @zhemao, we should definitely point to that.

MarcoGeuna, 2016-11-07:
Totally agree with @mwachs5: your example is really helpful @zhemao, thank you!!

closed 2016-12-01 by hcook