chipsalliance/rocket-chip #347: Jtag DTM Synchronizer Broken

mwachs5, 2016-09-27:
My last commit #342 screwed up the JTAG DTM.  My tests didn't have good clean targets. Fixing the issue now.

mwachs5, 2016-09-27:
Closed by mistake.

One quickly fixed problem was that one of the connections was missing. However, there is another problem in the DebugModule where the RAM address wraps around (not all the address bits are considered in the comparison, so for a 28 byte RAM, address 0x0 can be written by a write to address 0x8). 

This issue has always existed, but the OpenOCD implementation didn't attempt to use those values in general. But with the slower synchronizer, OpenOCD gets "busy" response on startup, which seems to screw it up. Talking with @timsifive to see what the right fix is.

timsifive, 2016-09-27:
This is a bug in OpenOCD. A debugger shouldn't assume anything about accesses to Debug RAM address that fall outside the range indicated as supported by dramsize in dminfo.

closed 2016-09-30 by mwachs5