chipsalliance/rocket-chip #351: building emulator error: make[1]: VTestHarness.mk: No such file or directory

ksh12, 2016-09-27:
Hi,

after getting the rocketchip git souces and running the build.sh and successfully completing it.
I did: 
cd emulator; 
make

This results in the following error:

_make[1]: VTestHarness.mk: No such file or directory
make[1]: *** No rule to make target `VTestHarness.mk'.  Stop.
make[1]: Leaving directory_

Can someone please point out what could be the cause of this?

Thanks.

zhemao, 2016-09-27:
That file is supposed to be generated by verilator. What do you see when you look inside the emulator/generated-src/rocketchip.DefaultConfig directory?

ksh12, 2016-09-28:
The emulator/generated-src/rocketchip.DefaultConfig dir is empty.

Although i do see the following files and dir in the emulator/generated-src dir:

DefaultConfig.cfg
DefaultConfig.cst
DefaultConfig.knb
consts.DefaultConfig.h
rocketchip.DefaultConfig
rocketchip.DefaultConfig.d
rocketchip.DefaultConfig.fir
rocketchip.DefaultConfig.prm
rocketchip.DefaultConfig.v

The error message ( not complete one, only the error related part) is below:

[info] [6.878] Done elaborating.
[success] Total time: 10 s, completed Sep 28, 2016 9:31:45 AM

echo "#ifndef __CONST_H__" > /user/ksh12/rocket-chip/emulator/generated-src/consts.DefaultConfig.h
echo "#define __CONST_H__" >> /user/ksh12/rocket-chip/emulator/generated-src/consts.DefaultConfig.h
sed -E 's/\(([A-Za-z0-9_]+),([A-Za-z0-9_]+)\)/#define \1 \2/' /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.prm >> /user/ksh12/rocket-chip/emulator/generated-src/consts.DefaultConfig.h
echo "#endif // __CONST_H__" >> /user/ksh12/rocket-chip/emulator/generated-src/consts.DefaultConfig.h
mkdir -p /user/ksh12/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /user/ksh12/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.fir -o /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v -X verilog
mkdir -p /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig
/user/ksh12/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert -Wno-STMTDLY --x-assign unique -I/user/ksh12/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/user/ksh12/install/include -I/user/ksh12/rocket-chip/csrc -DVERILATOR -include /user/ksh12/rocket-chip/csrc/verilator.h" -Mdir /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig \
        -o /user/ksh12/rocket-chip/emulator/emulator-TestHarness-DefaultConfig /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /user/ksh12/rocket-chip/csrc/emulator.cc /user/ksh12/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/user/ksh12/install/lib -Wl,-rpath,/user/ksh12/install/lib -L/user/ksh12/rocket-chip/emulator -lfesvr -lpthread" \
        -CFLAGS "-I/user/ksh12/rocket-chip/emulator/generated-src -include /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h -include /user/ksh12/rocket-chip/emulator/generated-src/consts.DefaultConfig.h"
make -C /user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig -f VTestHarness.mk
make[1]: Entering directory `/user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig'
make[1]: VTestHarness.mk: No such file or directory
make[1]: *** No rule to make target `VTestHarness.mk'.  Stop.
make[1]: Leaving directory `/user/ksh12/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig'
make: *** [emulator-TestHarness-DefaultConfig] Error 2

xfguo, 2016-10-02:
Hi @ksh12 , try delete verilator then rebuild it like this:

```
cd emualtor
rm -rf verilator
make 
```

ksh12, 2016-10-04:
Thanks xfguo.

closed 2016-10-04 by ksh12