chipsalliance/rocket-chip #358: Firrtl is called twice during RTL compilation

ben-k, 2016-09-28:
This can be observed in `vsim/` with a simple `make -n verilog`.  Chisel is run, and then firrtl is run twice with identical arguments.

Presumably there is some bug in `Makefrag-verilog` that is causing this.

zhemao, 2016-09-28:
Yeah, I'm getting something like this as well. Although for me, it's running VCS twice.

zhemao, 2016-09-28:
Okay, so the result of "make -n" is a bit deceptive. The issue is that the FIRRTL rule generates both the .v and .conf file. The first run of FIRRTL is for the .conf file and the second for the .v file. When running for real, the timestamps should update so that the second run is unnecessary. So for some reason one of the timestamps isn't being updated or a dependency is wrong.

terpstra, 2016-11-23:
This does not appear to happen when one actually runs make without '-n', even when -j 1000 is used.

ben-k, 2016-11-23:
Hm, I disagree.  I just cloned a clean rocket-chip; when I run `make verilog` in vsim it runs the following commands (in this sequence):

<pre>
mkdir -p /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/
cd /home/bkeller/work/morerocket/0-rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/bkeller/work/morerocket/0-rocket-chip/sbt-launch.jar "run-main rocketchip.Generator /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src rocketchip TestHarness rocketchip DefaultConfig"
mkdir -p /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/
<b>java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/bkeller/work/morerocket/0-rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.fir -o /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.conf</b>
cd /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src && \
rm -f /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v && \
/home/bkeller/work/morerocket/0-rocket-chip/vsim/vlsi_mem_gen /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.conf >> /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v.tmp && \
mv /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v.tmp /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v
echo "\`ifndef CONST_VH" > /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/consts.DefaultConfig.vh
echo "\`define CONST_VH" >> /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/consts.DefaultConfig.vh
sed -r 's/\(([A-Za-z0-9_]+),([A-Za-z0-9_]+)\)/`define \1 \2/' /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.prm >> /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/consts.DefaultConfig.vh
echo "\`endif // CONST_VH" >> /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/consts.DefaultConfig.vh
mkdir -p /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/
<b>java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/bkeller/work/morerocket/0-rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.fir -o /home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/bkeller/work/morerocket/0-rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.conf</b>
</pre>

I've bolded the double Firrtl call.  If I push `make -j verilog` I still get the redundancy, although the sequence is different.

colinschmidt, 2016-11-24:
As far as I can remember the problem is that you have two targets on the left hand side of the colon on https://github.com/ucb-bar/rocket-chip/blob/master/vsim/Makefrag-verilog#L15 You need to use % to get it to run once (the way the chisel call works). At some point in the past we changed to using ```$(long_name)``` but I'm not sure why

colinschmidt, 2016-11-24:
Things worked for me when I replaced ```$(long_name)``` with % so I just made that PR.

PR #452 (2016-11-25): Use % in makefrag-verilog to prevent double firrtl execution
Fixes #358
closed 2016-11-25 by terpstra