chipsalliance/rocket-chip #371: Get instruction access fault exception near trap vector in bbl

sashimi-yzh, 2016-10-02:
With the newest repo, including rocket-chip, bbl and linux (actually  I use the `priv-1.9` branch), I can run linux over the rocketchip FPGA implementation without FPU. But when I run the STREAM benchmark, which will perform floating point operations repeatedly, in linux, I get an instruction access fault exception near the trap vector in bbl.
```
../machine/mtrap.c:34: machine mode: unhandlable trap 1 @ 0x000000008000000c
```
Here are the disassembly result.
```
0000000080000000 <_ftext>:
    80000000:	2000006f          	j	80000200 <do_reset>

0000000080000004 <nmi_vector>:
    80000004:	28d0006f          	j	80000a90 <bad_trap>

0000000080000008 <trap_vector>:
    80000008:	34011173          	csrrw	sp,mscratch,sp
    8000000c:	1a010463          	beqz	sp,800001b4 <restore_regs+0x94>
    80000010:	04a13823          	sd	a0,80(sp)
    80000014:	04b13c23          	sd	a1,88(sp)
```

Actually, I have tried this with FPU as well as BOOM, and I get a similar exception. Sometimes the faulting address will be `0x80000008`. I guess this issue may be related to floating points.

But first of all, I am confused about in what kind of situations I will get an instruction access fault exception in machine mode.

Can anyone give me some hints? Thanks in advance!

sashimi-yzh, 2016-10-05:
It seems that commit 064c9ebdc633 fixes this issue. After applying the patch, I run STREAM benchmark repeatedly for about 3.5 hours, everything still works well.

sashimi-yzh, 2016-10-05:
After applying the same patch to BOOM, I run the same benchmark for about 3 hours. It still works well. :)

terpstra, 2016-11-23:
Since this seems to be fixed, I am closing this issue.

closed 2016-11-23 by terpstra