chipsalliance/rocket-chip #383: AHB spec signal hreadyout missing in Hasti

alonamid, 2016-10-05:
According to the AHB 3 spec, AHB slaves should have a signal called hreadyout to indicate that a transfer has finished on the bus. I don't see this signal (or a similar one) in the hasti implementation. Is this intentional or masked by a different signal?
@zhemao @terpstra

terpstra, 2016-10-05:
Sure there is:

  *val* hready = Bool(INPUT)

alonamid, 2016-10-05:
So where is the equivalent of a slave's hreadyin? (the global hready feedback)
http://mazsola.iit.uni-miskolc.hu/~drdani/docs_arm/IHI0033A_AMBA3_AHB_Lite.pdf
(page 55)

terpstra, 2016-10-06:
> IHI0033A_AMBA3_AHB_Lite.pdf

There is no 'hreadyin' in AHB. The used to be this signal in Hasti, because
the original author could not get the flow control to work without it. The
new Hasti Xbar does not have this limitation and so the non-standard signal
was removed.

'hreadyout' and 'hready' in the spec are the same signal. They just changed
the name of the signal after going through the Xbar. hreadyout goes from
slave to Xbar and hready from Xbar to master. There is no hready that is an
INPUT to the slave.

alonamid, 2016-10-06:
Every AHB reference that I can find online does mention an hready signal as an input to the slave so that the slave can determine when the previously selected slave has completed its final transfer.
As far as I can tell it looks like it’s in the standard (http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0243c/I1008569.html)

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/482.html

https://community.arm.com/thread/2996

http://eece.cu.edu.eg/~akhattab/files/courses/ca/AHB_Signals.pdf

terpstra, 2016-10-06:
> doc.ddi0243c/I1008569.html)

Hmm. Interesting. Well, feel free to drive the slave hreadyin with a
constant true, then.

terpstra, 2016-10-06:
Thank you for the links and the patience. It seems that this signal was missing in the AHB specification I read and was added in a later revision! The existence of hreadyin certainly makes the job of the interconnect easier, as shown in Figure 4-2 of the 0033A revision. In any case, The Hasti interconnect inserts a pipeline bubble when switching between slaves to avoid the problem hreadyin solves. So just set hreadyin to high and it will work just fine.

When I revisit AHB later this month, I might simplify the interconnect by sending a hready to the slaves.

terpstra, 2016-10-06:
Or rather, tie the slave's hreadyin to the slave's hreadyout.

PR #485 (2016-12-09): Diplomatic AHB
This adds a hopefully more reliable AHB adapter. This closes #383.
closed 2016-12-09 by hcook