chipsalliance/rocket-chip #401: Rocket comes out of reset with valid icache data

colinschmidt, 2016-10-16:
Certain randomization values can lead to rocket's icache immediately asserting a valid response. In default rocket-chip configurations this doesn't have any affect on the processor because a debug interrupt comes before the processor is ready to begin executing. But if you were to use a tethered interface to bring up the system there would be no need for the debug interrupt at the start.

My issue then becomes a question:
Should I setup mtvec or similar mechanism before the processor is reset? 
Can I rely on the first instruction at the reset vector to be fetched correctly? 

If this is simply a bug my current fix to this issue for me is the following diff in icache.scala

```
-      val s2_hit = RegEnable(s1_hit, !stall)
+      val s2_hit = RegEnable(s1_hit, Bool(false), !stall)
```

I think @aswaterman would be the best one to answer this question.

mwachs5, 2016-10-17:
How many ticks of clock are you asserting with reset high (need to assert several to "flush" the pipeline)?

colinschmidt, 2016-10-17:
Probably a couple hundred thousand. In my environment the coreplex comes out of reset first and then the program is loaded via a tether, after all that is done rocket's reset is lowered. So its definitely in reset for long enough. Let me get you a seed and hash of rocket-chip that presents the problem for me.

colinschmidt, 2016-10-17:
I used the random_seed 894732643 in vcs with the rocket-chip hash ee66fd28eba3f6a9479d209240dc7fb33d11c8a8

colinschmidt, 2016-10-17:
For the whole time reset is asserted you should see that the icache is asserting valid.

mwachs5, 2016-10-17:
I'll try to reproduce but think I need more than just the seed and hash... would your CONFIG/testbench impact the randomization?

colinschmidt, 2016-10-17:
This is clean rocket-chip on that hash with the default config

mwachs5, 2016-10-18:
Thanks for the pointer. I was able to reproduce the issue, but agree that @aswaterman is best to approve the fix.

aswaterman, 2016-10-18:
Yes, that looks good to me.

mwachs5, 2016-10-18:
@colinschmidt do you want to make  PR against master with this fix?

closed 2016-10-22 by colinschmidt