chipsalliance/rocket-chip #405: verilog load : dc_shell

samiasd, 2016-10-18:
dear all,,
I tried to load the verilog file into dc_shell and I got the following error
dc_shell> read_verilog /rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.v
******************
*****************
****************
Error:  /rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.v:180288: The width (2147483647 ) is too big.  (ELAB-1012)
*** Presto compilation terminated with 1 errors. ***
Error: Can't read 'verilog' file '/home/salamin/rocket/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.v'. (UID-59)
No designs were read
dc_shell> 

any suggestion please
Regards

mwachs5, 2016-10-18:
What do you see on line 180288 of the verilog file?

kammoh, 2016-10-18:
Hmm, strange. Never seen that during several runs of DC. And my current
master checkout of rocket-chip with the default DefaultConfig results in
176734 lines of verilog.

samiasd, 2016-10-18:
@mwachs5 at line 180288 ,,, it is the start of new module as followed
module LatencyPipe(
  input   clock,
  input   reset,
  output  io_in_ready,
  input   io_in_valid,
  input  [1:0] io_in_bits_resp,
  input  [63:0] io_in_bits_data,
  input   io_in_bits_last,
  input  [4:0] io_in_bits_id,
  input   io_in_bits_user,
  input   io_out_ready,
  output  io_out_valid,
  output [1:0] io_out_bits_resp,
  output [63:0] io_out_bits_data,
  output  io_out_bits_last,
  output [4:0] io_out_bits_id,
  output  io_out_bits_user
);

@kammoh could you please share your configuration or tcl file??
thanx

samiasd, 2016-10-20:
No answer??

closed 2016-10-20 by samiasd