chipsalliance/rocket-chip #408: Is io.resp.data in rocket/icache.scala even used?

jackkoenig, 2016-10-20:
From
```scala
class ICacheResp(implicit p: Parameters) extends CoreBundle()(p) with HasL1CacheParameters {
  val data = Bits(width = coreInstBits)
  val datablock = Bits(width = rowBits)
}
```

As far as I can tell, only datablock is used.

ccelio, 2016-10-20:
Is it not? I've never seen datablock before, but I use data in BOOM.

https://github.com/ucb-bar/riscv-boom/blob/master/src/main/scala/fetch.scala#L129

colinschmidt, 2016-10-20:
Hwacha uses datablock

colinschmidt, 2016-10-20:
@ccelio I think you are instantiating the frontend not the ICache itself so thats why you use data and not datablock. Hwacha has its own frontend and so instantiates the actual ICache to use its datablock.

ccelio, 2016-10-20:
Oh, good point.

closed 2016-11-26 by aswaterman